fkie/multimaster_fkie #16: master discovery: protocol not available error on Odroid U3

richardw347, 2015-01-13:
I'm trying to get a multimaster setup working on an Odroid U3 and get and error when running master_discovery. Here is the screen dump, any help would be appreciated:

rosrun master_discovery_fkie master_discovery
[DEBUG] [WallTime: 1421188855.609521] init_node, name[/master_discovery], pid[2550]
[DEBUG] [WallTime: 1421188855.610482] binding to 0.0.0.0 0
[DEBUG] [WallTime: 1421188855.611127] bound to 0.0.0.0 51241
[DEBUG] [WallTime: 1421188855.612483] ... service URL is rosrpc://odroid:51241
[DEBUG] [WallTime: 1421188855.613140] [/master_discovery/get_loggers]: new Service instance
[DEBUG] [WallTime: 1421188855.617563] ... service URL is rosrpc://odroid:51241
[DEBUG] [WallTime: 1421188855.618277] [/master_discovery/set_logger_level]: new Service instance
[INFO] [WallTime: 1421188855.636224] ROS Master URI: http://odroid:11311
[INFO] [WallTime: 1421188855.667103] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1421188855.667809] Heart beat [Hz]: 2
[INFO] [WallTime: 1421188855.668324] Static hosts: []
[INFO] [WallTime: 1421188855.668809] Approx. network load: 136 bytes/s
/opt/ros/indigo/lib/python2.7/dist-packages/master_discovery_fkie/master_discovery.py:353: SyntaxWarning: The publisher should be created with an explicit keyword argument 'queue_size'. Please see http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers for more information.
  self.pubstats = rospy.Publisher("~linkstats", LinkStatesStamped)
[INFO] [WallTime: 1421188855.710610] Start broadcasting at ('226.0.0.0', 11511)
[INFO] [WallTime: 1421188855.711547] Init multicast socket
[ERROR] [WallTime: 1421188855.712425] Error while start master_discovery: [Errno 92] Protocol not available
Killed

richardw347, 2015-01-13:
Some more details on the setup, I'm running Ubuntu 14.04 + ROS Indigo on the odroid. I've installed multimaster_fkie from the debs and the indigo-devel branch from github and the result is the same.

Also zeroconf work with not problems so I'm guessing it might be a udp multicast issue with the odroid kernel but I'm not familiar enough with that way master_discovery work to say for sure.

Thanks

atiderko, 2015-01-14:
Do you use IPv4 or IPv6. It seems, there is no IPv4 configured. Try to configure IPv4, because in debs there was a bug with IPv6. Or you can use the multimaster_fkie from github. Then you must set the parameter `mcast_group` to IPv6 multicast group (for example: `ff02::15`) while start of master_discovery.

regards,
Alexander

atiderko, 2015-02-03:
Does it work?

ChongYunPeng, 2016-07-21:
Hi atiderko,

I am using Odroid U3 + Ubuntu 16.04 + ROS kinetic and I am facing the same issue as richard.

[INFO] [WallTime: 1469069117.619408] ROS Master URI: http://localhost:11311
[INFO] [WallTime: 1469069117.656470] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1469069117.657304] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1469069117.657977] Active request after [sec]: 60
[INFO] [WallTime: 1469069117.658594] Remove after [sec]: 300
[INFO] [WallTime: 1469069117.659197] Robot hosts: []
[INFO] [WallTime: 1469069117.659814] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1469069117.699024] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1469069117.700656] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1469069117.712871] Listen for multicast at ('226.0.0.0', 11511)
[ERROR] [WallTime: 1469069117.713850] Error while start master_discovery: [Errno 92] Protocol not available
Killed

I have tried both the master and  the kinetic-devel branch, built from source since the kinetic package is not available yet.

However when I tried using Odroid XU4 + Ubuntu 16.04 + ROS kinetic, the run was successful. Do you think what the issue may be?

Thanks

atiderko, 2016-07-21:
Is the Multicast support enabled in Odroid?
Please, try to start the multicast_discovery with `_send_mcast:=False` option.

ChongYunPeng, 2016-07-21:
Below is the error:
$rosrun master_discovery_fkie master_discovery

[INFO] [WallTime: 1469082276.217958] ROS Master URI: http://odroidserver1:11311
[WARN] [WallTime: 1469082276.253567] This master_discovery is invisible because it send no heart beat messages! Set ~send_mcast to true or add hosts to ~robot_hosts.
[WARN] [WallTime: 1469082276.254463] Multicast is disabled. Use ~active_request_after(60.00) ot set ~heartbeat_hz to new value: 0.0167
[INFO] [WallTime: 1469082276.255150] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1469082276.255768] Heart beat [Hz]: 0.0166666666667
[INFO] [WallTime: 1469082276.256365] Active request after [sec]: 60
[INFO] [WallTime: 1469082276.256959] Remove after [sec]: 300
[INFO] [WallTime: 1469082276.257626] Robot hosts: []
[INFO] [WallTime: 1469082276.258264] Approx. mininum avg. network load: 0.00 bytes/s
[INFO] [WallTime: 1469082276.297719] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1469082276.299406] Subscribe to parameter `/roslaunch/uris`
[WARN] [WallTime: 1469082276.311552] Multicast disabled!
[INFO] [WallTime: 1469082276.312414] + Bind to unicast socket @(192.168.4.194:11511)
[ERROR] [WallTime: 1469082276.313329] Error while start master_discovery: [Errno 92] Protocol not available
Killed

I think I did enable multicast.
$cat /proc/sys/net/ipv4/icmp_echo_ignore_broadcasts 
0

atiderko, 2016-07-21:
Please try:
`rosrun master_discovery_fkie master_discovery _send_mcast:=False`

ChongYunPeng, 2016-07-21:
It gives the same error message.

atiderko, 2016-07-21:
:-/ ok, next try:
please checkout the `issue16` branch and run it again. I added some more outputs.

ChongYunPeng, 2016-07-21:
Below is the error message:

$ rosrun master_discovery_fkie master_discovery _send_mcast:=False

[INFO] [WallTime: 1469084272.560516] ROS Master URI: http://odroidserver1:11311
[WARN] [WallTime: 1469084272.596066] This master_discovery is invisible because it send no heart beat messages! Set ~send_mcast to true or add hosts to ~robot_hosts.
[WARN] [WallTime: 1469084272.596950] Multicast is disabled. Use ~active_request_after(60.00) ot set ~heartbeat_hz to new value: 0.0167
[INFO] [WallTime: 1469084272.597698] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1469084272.598340] Heart beat [Hz]: 0.0166666666667
[INFO] [WallTime: 1469084272.598933] Active request after [sec]: 60
[INFO] [WallTime: 1469084272.599513] Remove after [sec]: 300
[INFO] [WallTime: 1469084272.600091] Robot hosts: []
[INFO] [WallTime: 1469084272.600665] Approx. mininum avg. network load: 0.00 bytes/s
[INFO] [WallTime: 1469084272.640271] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1469084272.641908] Subscribe to parameter `/roslaunch/uris`
[WARN] [WallTime: 1469084272.654203] Multicast disabled!
[INFO] [WallTime: 1469084272.655077] + Bind to unicast socket @(192.168.4.194:11511)
[INFO] [WallTime: 1469084272.655793]   addrinfo: (2, 1, 6, '', ('192.168.4.194', 0))
[INFO] [WallTime: 1469084272.656465]   set socketopt SO_REUSEPORT
[ERROR] [WallTime: 1469084272.657162] Error while start master_discovery: [Errno 92] Protocol not available
Traceback (most recent call last):
  File "<string>", line 104, in main
  File "/home/hope/catkin_ws/src/master_discovery_fkie/src/master_discovery_fkie/master_discovery.py", line 558, in __init__
    self._init_socket(True)
  File "/home/hope/catkin_ws/src/master_discovery_fkie/src/master_discovery_fkie/master_discovery.py", line 588, in _init_socket
    self.socket = DiscoverSocket(self.mcast_port, self.mcast_group, unicast_only=(not self._send_mcast))
  File "/home/hope/catkin_ws/src/master_discovery_fkie/src/master_discovery_fkie/udp.py", line 105, in __init__
    self.unicast_socket = UcastSocket('', port)
  File "/home/hope/catkin_ws/src/master_discovery_fkie/src/master_discovery_fkie/udp.py", line 405, in __init__
    self.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 92] Protocol not available

Killed

This looks like the socket error that is discussed in this link:
http://stackoverflow.com/questions/37514180/python-server-socket-error-errno-92-protocol-not-available

atiderko, 2016-07-21:
I added a try... catch block. Please try it again.

ChongYunPeng, 2016-07-21:
$ rosrun master_discovery_fkie master_discovery _send_mcast:=False

[INFO] [WallTime: 1469087465.012649] ROS Master URI: http://odroidserver1:11311
[WARN] [WallTime: 1469087465.046711] This master_discovery is invisible because it send no heart beat messages! Set ~send_mcast to true or add hosts to ~robot_hosts.
[WARN] [WallTime: 1469087465.047571] Multicast is disabled. Use ~active_request_after(60.00) ot set ~heartbeat_hz to new value: 0.0167
[INFO] [WallTime: 1469087465.048335] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1469087465.048965] Heart beat [Hz]: 0.0166666666667
[INFO] [WallTime: 1469087465.049567] Active request after [sec]: 60
[INFO] [WallTime: 1469087465.050155] Remove after [sec]: 300
[INFO] [WallTime: 1469087465.050728] Robot hosts: []
[INFO] [WallTime: 1469087465.051301] Approx. mininum avg. network load: 0.00 bytes/s
[INFO] [WallTime: 1469087465.091573] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1469087465.093196] Subscribe to parameter `/roslaunch/uris`
[WARN] [WallTime: 1469087465.105301] Multicast disabled!
[INFO] [WallTime: 1469087465.106485] + Bind to unicast socket @(192.168.4.194:11511)
[INFO] [WallTime: 1469087465.107488]   addrinfo: (2, 1, 6, '', ('192.168.4.194', 0))
[INFO] [WallTime: 1469087465.108490]   set socketopt SO_REUSEPORT
[WARN] [WallTime: 1469087465.109435] can not set SO_REUSEPORT option to the socket
[INFO] [WallTime: 1469087465.251340] Detected master discovery: http://192.168.4.194:11611
[INFO] [WallTime: 1469087465.356498] Added master with ROS_MASTER_URI=http://odroidserver1:11311/

There is no error when this is run. However the error still occurs when _send_mcast:=True

May I know if it is because of the code or the platform? I do not face this problem when I am running on odroid XU4.

Thank you for your help so far.

atiderko, 2016-07-21:
It fails while setting the `SO_REUSEPORT` option for the socket. I think it is because the Ubuntu/kernel running on the platform.
In the previous version I forgot to add a try... catch block while set the option for multicast socket. I did it now. Please test it last time. It should run, but some functions are not available (multiple instances of master_discovery for simulation or discovering with node_manager on the same host...)
If it works, I will put this fix into the main branch.
Thank you very much for testing!!!

ChongYunPeng, 2016-07-21:
Now when I run both the following commands there are no errors, but there are some warning messages.

`$ rosrun master_discovery_fkie master_discovery _send_mcast:=False
`
[INFO] [WallTime: 1469090351.779601] ROS Master URI: http://odroidserver1:11311
[WARN] [WallTime: 1469090351.813445] This master_discovery is invisible because it send no heart beat messages! Set ~send_mcast to true or add hosts to ~robot_hosts.
[WARN] [WallTime: 1469090351.814318] Multicast is disabled. Use ~active_request_after(60.00) ot set ~heartbeat_hz to new value: 0.0167
[INFO] [WallTime: 1469090351.815008] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1469090351.815640] Heart beat [Hz]: 0.0166666666667
[INFO] [WallTime: 1469090351.816226] Active request after [sec]: 60
[INFO] [WallTime: 1469090351.816796] Remove after [sec]: 300
[INFO] [WallTime: 1469090351.817374] Robot hosts: []
[INFO] [WallTime: 1469090351.818005] Approx. mininum avg. network load: 0.00 bytes/s
[INFO] [WallTime: 1469090351.857396] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1469090351.859048] Subscribe to parameter `/roslaunch/uris`
[WARN] [WallTime: 1469090351.871163] Multicast disabled!
[INFO] [WallTime: 1469090351.872022] + Bind to unicast socket @(192.168.4.194:11511)
[WARN] [WallTime: 1469090351.872846] SO_REUSEPORT failed: Protocol not available, some functions are not available.
[INFO] [WallTime: 1469090352.014382] Detected master discovery: http://192.168.4.194:11611
[INFO] [WallTime: 1469090352.119707] Added master with ROS_MASTER_URI=http://odroidserver1:11311/

`$ rosrun master_discovery_fkie master_discovery _send_mcast:=True
`                        
[INFO] [WallTime: 1469090426.424088] ROS Master URI: http://odroidserver1:11311
[INFO] [WallTime: 1469090426.459402] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1469090426.460247] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1469090426.460905] Active request after [sec]: 60
[INFO] [WallTime: 1469090426.461516] Remove after [sec]: 300
[INFO] [WallTime: 1469090426.462128] Robot hosts: []
[INFO] [WallTime: 1469090426.462752] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1469090426.502255] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1469090426.503926] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1469090426.516137] Listen for multicast at ('224.0.0.1', 11511)
[WARN] [WallTime: 1469090426.517051] SO_REUSEPORT failed: Protocol not available, some functions are not available.
[INFO] [WallTime: 1469090426.680409] Detected master discovery: http://192.168.4.194:11611
[INFO] [WallTime: 1469090426.685397] Detected master discovery: http://192.168.4.245:11611
[INFO] [WallTime: 1469090426.688051] Detected master discovery: http://192.168.4.202:11611
[INFO] [WallTime: 1469090426.788091] Added master with ROS_MASTER_URI=http://odroidserver1:11311/
[INFO] [WallTime: 1469090426.796118] Added master with ROS_MASTER_URI=http://jason:11311/
[WARN] [WallTime: 1469090426.816035] Master discovered with not known hostname ROS_MASTER_URI:='http://odroidserver2:11311/'. Fix your network settings!

atiderko, 2016-07-21:
Ok, now you can use `master_discovery` with default `_send_mcast` value (True).
The warning in the last line says what `odroidserver2` can not be resolved on `odroidserver1`. Perhaps you have to add this host to the `/etc/hosts`

ChongYunPeng, 2016-07-21:
Hey you are right. I forgot to add odroidserver2 in /etc/hosts.

Thanks a lot. It's working now.

atiderko, 2016-07-21:
Thank you for your help!

closed 2016-07-21 by atiderko

AlexisTM, 2018-04-04:
For future users: You can use `hostname.local` instead of hostname if you have the avahi zeroconf daemon installed.