fkie/multimaster_fkie #19: bash: rosrun: command not found issue

skill-io, 2015-01-29:
hi,

i know this is already covered in the troubleshooting section but the proposed solution sadly does not work in my case (both machines run ubuntu 14.04, indigo)

    To run a node on remote host, an SSH connection will be established without setting any enviroment variables. Try to comment out in .bashrc: 

    #[ -z "$PS1" ] && return

i do not have such an entry on my machine or on our robot. also adding this line does not make a difference. is there something else i can try?

thx in advance

atiderko, 2015-01-29:
hi,

in 14.04 this line looks different. Try to put all `source ...` and ROS setup lines on top of the .bashrc.

regards

skill-io, 2015-01-29:
thx a lot for the ultra quick reply!

placing the ROS setup commands on top of the .bashrc solved the problem. 

works like a charm now! 
thank you very much

closed 2015-02-03 by atiderko