fkie/multimaster_fkie #20: master discovery is not working on os x

burrimi, 2015-02-06:
When I start master discovery on os x I get the following error:

[WARN] 06.02.2015 17:54:53.423242 (master_monitor.py:_limited_log:651):
MasterMonitor[/master_discovery]: can't get PID: [Errno 61] Connection refused

atiderko, 2015-02-06:
Are there other errors or warnings? Does the master_discovery crash?

burrimi, 2015-02-06:
[DEBUG] [WallTime: 1423253710.785981] init_node, name[/node_manager], pid[92640]
[DEBUG] [WallTime: 1423253710.786370] binding to 0.0.0.0 0
[DEBUG] [WallTime: 1423253710.786631] bound to 0.0.0.0 55393
[DEBUG] [WallTime: 1423253710.787057] ... service URL is rosrpc://mac.asl_mav:55393
[DEBUG] [WallTime: 1423253710.787242] [/node_manager/get_loggers]: new Service instance
[DEBUG] [WallTime: 1423253710.789027] ... service URL is rosrpc://mac.asl_mav:55393
[DEBUG] [WallTime: 1423253710.789213] [/node_manager/set_logger_level]: new Service instance
[INFO] [WallTime: 1423253711.108096] listen for logs on /rosout
[DEBUG] [WallTime: 1423253711.164328] connecting to mac.asl_mav 55393
[INFO] [WallTime: 1423253711.748395] Start RPC-XML Server at ('0.0.0.0', 22622)
[INFO] [WallTime: 1423253711.748881] Subscribe to parameter `/roslaunch/uris`
[DEBUG] [WallTime: 1423253713.740936] MASTERINFO from mac.asl_mav (http://mac.asl_mav:11311/) received
2015-02-06 21:15:24.949 Python[92640:d0b] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
[INFO] [WallTime: 1423253725.203563] Run without config: /usr/bin/screen -c /Users/username/.ros/log/_master_discovery.conf -L -dmS _master_discovery /Users/username/ros/catkin_ws/src/multimaster_fkie/master_discovery_fkie/nodes/master_discovery _mcast_port:=11511 _send_mcast:=True _static_hosts:=[] __name:=master_discovery
[DEBUG] [WallTime: 1423253725.860580] Unable to connect subscriber to publisher [http://mac.asl_mav:55501/] for topic [/rosout]: unable to requestTopic: [Errno 61] Connection refused
[WARN] [WallTime: 1423253726.263804] MasterMonitor[/master_discovery]: can't get PID: [Errno 61] Connection refused
[DEBUG] [WallTime: 1423253726.943554] MASTERINFO from mac.asl_mav (http://mac.asl_mav:11311/) received
[DEBUG] [WallTime: 1423253743.284028] MASTERINFO from mac.asl_mav (http://mac.asl_mav:11311/) received

burrimi, 2015-02-06:
this is what I get, when I just run the master_discovery.
user@hostname$ rosrun master_discovery_fkie master_discovery 
[DEBUG] [WallTime: 1423253946.321723] init_node, name[/master_discovery], pid[95068]
[DEBUG] [WallTime: 1423253946.322053] binding to 0.0.0.0 0
[DEBUG] [WallTime: 1423253946.322267] bound to 0.0.0.0 56040
[DEBUG] [WallTime: 1423253946.322625] ... service URL is rosrpc://mac.asl_mav:56040
[DEBUG] [WallTime: 1423253946.322760] [/master_discovery/get_loggers]: new Service instance
[DEBUG] [WallTime: 1423253946.348085] ... service URL is rosrpc://mac.asl_mav:56040
[DEBUG] [WallTime: 1423253946.348316] [/master_discovery/set_logger_level]: new Service instance
[INFO] [WallTime: 1423253946.356265] ROS Master URI: http://localhost:11311
[INFO] [WallTime: 1423253946.370294] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1423253946.370541] Heart beat [Hz]: 2
[INFO] [WallTime: 1423253946.370706] Static hosts: []
[INFO] [WallTime: 1423253946.370855] Approx. network load: 136 bytes/s
[INFO] [WallTime: 1423253946.372759] Start broadcasting at ('226.0.0.0', 11511)
[INFO] [WallTime: 1423253946.372925] Init multicast socket
[ERROR] [WallTime: 1423253946.392381] Error while start master_discovery: [Errno 6] Device not configured
Killed: 9

atiderko, 2015-02-07:
the master_discovery dies at startup, therefore the error: `can't get PID: [Errno 61] Connection refused`.
I don't have an OS X, so I can't reproduce the error. Is the network device configured for IPv4 multicast communication?

atiderko, 2017-01-26:
we have test the master_discovery on a macOS and fixed some issues. It should now work.

closed 2017-01-26 by atiderko