fkie/multimaster_fkie #22: Dependency on python-qt

fmina, 2015-05-21:
Hi, 
When we start to discover the master on remote host we get python error stating that the python-qt-buinding module is not existing. However we have limited storage on the robot and we want to avoid installing python-qt. Is the dependency on python-qt necessary even on the robot?
We tried to start only master-discovery on the robot and the node_manager discovered the robot, but same issue happens when we start a node for instance.

Thank you

atiderko, 2015-05-22:
Hi,
to run nodes on remote hosts from node_manager we use scripts in scripts folder of node_manager package. These scripts use the code of node_manager to simplify the maintenance. It is possible to extract the code, so it does not need the python-qt. I will try to create this script, but next week.
Regards

atiderko, 2015-05-22:
hi,
I created a backup script, which work without qt bindings. It was not a lot of changes :-)
Try it out, it is in master branch.

fmina, 2015-05-22:
Thanks, I will try it and let you know :)

fmina, 2015-05-26:
Works perfectly, please close the issue. 
Thanks :+1:

closed 2015-05-27 by atiderko