fkie/multimaster_fkie #23: Some master_sync parameter don't seem to be working.

KevinOlsthoorn, 2015-10-16:
Hi,

Whenever I try to use the ignore_topics or sync_topics parameter, it doesn't seem to have any effect.
The terminal is indicating that the parameters are entered correctly by printing them but all the existing topics are received by other masters in the same network regardless of the entered parameter values.
I have taken a look a look at the Hyrdo (which we are using) branch implementation and I couldn't find any implementation regarding these parameters.
I was only able to find some implementation about the sync_topics_on_demand parameter but not about the others.

Thanks.

atiderko, 2015-10-16:
Hi,

sync_parameter handling is implemented in `master_discovery_fkie/filter_interface.py`.
It seems what you try to hide some topics, so remote masters can not retrieve this info. This does not work! You have to setup the filter on masters which requests the topic information.

Regars

KevinOlsthoorn, 2015-10-22:
Thanks for your reply, when I take a look at the ROS documentation about the master_sync_fkie it tells me that the ~sync_topics parameter is part of the master_sync_fkie node: http://wiki.ros.org/master_sync_fkie?distro=hydro.
Could you point me in the general direction on how to implement only receiving specific topics on remote masters?

KevinOlsthoorn, 2015-10-22:
Alright, I seem to have understood correctly. I added the sync_topics parameter to the remote masters and it seems to be working now. Now I was asking myself, since the remote master is specifying which topics to sync, does the broadcasting master send all the topics anyway and the remote masters filters these based on the given parameter values? Or does the remote master request only the selected topics from the broadcasting master to prevent all the topics from being send to remote masters over the network even though they might not want to receive them.

So I would like to know if specifying the sync_topics parameter on the remote masters will result in less traffic on the network.

Thanks.

atiderko, 2015-10-22:
Yes, you have understood correctly. If you specify sync_topics parameter on the remote master it will result in less network trafic. The master_sync node requests only specified topics from master_discovery.
The master_discovery node broadcasts only small messages with current timestamp of last change of ROS Master. The remote master_sync nodes compare the timestamps and request then the newest ROS Master state from master_discovery node. The idea is described on http://fkie.github.io/multimaster_fkie/

The ~sync_topics parameter is a part of the master_sync_fkie node, but it is a part of a filter, which is also used and implemented in master_discovery_fkie (for filtered requests).

Regards

PS: can I close this issue or do you have any questions?

KevinOlsthoorn, 2015-10-22:
Thanks for your explanation!

Everything is a lot more clear now.

You can close the issue if you want.

closed 2015-10-22 by atiderko