fkie/multimaster_fkie #26: Start multimaster-fkie using a remote master (ROS_MASTER_URI)

KevinOlsthoorn, 2015-11-27:
Hi,

We are trying to start the master-discovery and master-sync nodes on a device with a remote ROS_MASTER_URI.
But when we try this, the master-sync node tells us: 
"Master_discovery node appear not to running. Wait for topic with type 'MasterState."

When we take a look at the node manager we see we get another kind of error where it's complaining about the fact that the ROS_MASTER_URI is different than the origin IP address.

How can we start the discovery and sync nodes using a remote master?

atiderko, 2015-11-27:
Hi,

since the master_discovery updates the state of ROS Master by polling it is not a good idea to run ROS Master and master_discovery on different hosts.

If you still want to get it works, you need to comment out `and hostFromUri(val) == hostFromUri(masteruri)` @ https://github.com/fkie/multimaster_fkie/blob/master/master_discovery_fkie/src/master_discovery_fkie/interface_finder.py#L145 and https://github.com/fkie/multimaster_fkie/blob/master/master_discovery_fkie/src/master_discovery_fkie/interface_finder.py#L231

Regards,
atiderko

KevinOlsthoorn, 2015-11-27:
Thanks,

It's only going to be used for testing purposes.

I tried the changes and the warnings while running the master-sync-fkie node now seem to be gone.
Unfortunately I still have the same error in the node manager.
I tried changing the source code on both masters.
It seems like I am not yet able to broadcast topics to the other master.

atiderko, 2015-11-30:
Hi,

the master_discovery shows this warning, because in some cases a host has more then one interface and it can be an indication why something goes wrong.

In your case you have to ignore this warning.

In your problem: 
* are the master in the same network? 
* does the multicast work between these hosts? Try to set `~robot_hosts` parameter of master_discovery to other host. (for details, see the launch file of the master_discovery_fkie)
* are the topics synchronized? Perhaps it helps to set `~sync_remote_nodes` of master_sync_fkie to True.

regards,
atiderko

KevinOlsthoorn, 2015-12-03:
Hi,

You were right. We did seem to be able to broadcast correctly now!
We will just ignore the warning for now.

You can close the issue if you like.

Thanks.

closed 2015-12-03 by atiderko