fkie/multimaster_fkie #27: sync_topics parameter not working?

ZdenekM, 2015-12-03:
It looks like sync_topics is ignored (at least in my case). With following setup:

ROS1 (ROS_MASTER_URI=http://localhost:11312)
 - rostopic pub /test/topic std_msgs/String 'Hello World' -r 1
 - rostopic pub /test/topic_secret std_msgs/String 'Hello World!' -r 1
 - rosrun master_discovery_fkie master_discovery
 - rosrun master_sync_fkie master_sync _sync_topics:=["/test/topic"] _sync_topics_on_demand:=false

ROS2 (ROS_MASTER_URI=http://localhost:11311)
 - rosrun master_discovery_fkie master_discovery
 - rosrun master_sync_fkie master_sync _sync_topics_on_demand:=false
 - rostopic echo /test/topic -> works (as expected)
 - rostopic echo /test/topic_secret -> also works

Am I missing something or is it a bug? I would like to "flip" only selected topics between ROS1 and ROS2. Thanks in advance.

atiderko, 2015-12-03:
Hi ZdenekM,

you have to run `rosrun master_sync_fkie master_sync _sync_topics:=["/test/topic"]` on ROS2, too.
In your case you don't need to start master_sync on ROS1.

The parameter of master_sync specify which remote topics, services or nodes are registered/filtered by own ROS Master.  It does not matter if the master_sync is running on another host.

regards,
atiderko

ZdenekM, 2015-12-03:
Now it works fine. Thanks. This could be probably mentioned on wiki. Btw, is it possible to "remap" topic - I mean to have 'topic1' on ROS1 as 'topic2' to ROS2?

closed 2015-12-03 by ZdenekM

atiderko, 2015-12-03:
I am pleased :-). 

No, it is no possible to to rename topics while synchronization, because master_sync does not so deeply change ROS. There no new connections are created. master_sync registers only a new topic, the connections is done by ROS Master itself.