fkie/multimaster_fkie #28: using multimaster on machine with lan and wifi

Rimgaudas, 2015-12-18:
there is robot with board A and B
roscore running on A
multimaster runs on B
A and B connected via Lan 
B has a wifi 
even running core on B doesn't solve a problem.
is there a setting to choose which connection multimaster use?
there is also possibility that multicast is disabled on the network. How can I test it?

atiderko, 2015-12-18:
Hi,

1. you have to run roscore on A and B
2. you have to run `master_discovery` (a part of multimaster) on A and B
  - you can start it using `node_manager` (a part of multimaster)
3. if muticast is not supported by network `master_discovery` dies on start with an error
  - you can disable the usage of multicast by `send_mcast`
  - and if it starts but no other `master_discovery` nodes are discovered you can use `robot_hosts` to enter host to communicate with by unicast
4. You can bind `master_discovery` to an interface by `interface` or if you start `master_discovery` by `node_manager` add to the group the interface separated by `@` e.g. `226.0.0.0@192.168.0.20`

for other `master_discovery` parameters see the launch file in the package.

regards

Rimgaudas, 2016-01-20:
I tried to run it on two laptops
LAN works 
wifi doesn't 

I am able to:
  -ping to each other
  -ping to 224.0.0.1
  -master_discovery doesn't die  on start with an error 
so it must be some kind of network settings I guess
any idea how to test what is it ?

atiderko, 2016-01-21:
you can try it with `wireshark` and look for heartbeat messages of `master_discovery` on `udp.port ==11511` . Does receive each host from all other hosts these messages?

mwswartwout, 2016-02-23:
I am facing a similar issue to this. I am trying to set this up over WiFi, however I do not control the network I am on and I believe that something with their security or NAT may be interfering with multicast. Can you explain more how the `robot_hosts` parameter should work?

I am enable to ping by IP and hostname from machine A to B. What I have done is run, on machine A:
`rosrun master_discovery_fkie master_discovery robot_hosts=MachineBHostname`

And the opposite on Machine B. However they still do not discover each other. Is this the correct usage of `robot_hosts`?

atiderko, 2016-02-24:
Hi @mwswartwout,

the right command line to start on machine A:
`rosrun master_discovery_fkie master_discovery _robot_hosts:=[MachineBHostname] _send_mcast:=False`

on machine B:
`rosrun master_discovery_fkie master_discovery _robot_hosts:=[MachineAHostname] _send_mcast:=False`

if it does not work, try to start with additional parameter `_log_level:=DEBUG`. (You have to start master_discovery a second time to see the debug output). Perhaps you see what is going wrong or you post the log here.

The communication should work even if `send_mcast` is `False`.

regards,
atiderko

closed 2016-03-24 by Rimgaudas