fkie/multimaster_fkie #34: node manager warning

Georacer, 2016-03-30:
Hello,

I am trying to connect two roscores, one running on an Ubuntu laptop and another running on an Ubuntu Odroid XU3, both with ROS Indigo.

I have successfully followed all the steps for the single network configuration from here: http://www.iri.upc.edu/files/scidoc/1607-Multi-master-ROS-systems.pdf

I am able to publish a topic from the laptop and read it on the Odroid.

However, when I try to open the node_manager to get a better idea of my network, I get the following warning, regarding the quality of the link and the time slip.

[image]

Is there something I need to worry about or something I can do about it?

Thanks

atiderko, 2016-03-31:
Hello @Georacer,

the warning about time slip is particularly interesting when tf is synchronized. You can remove the warning by:
1. Increasing the `Max time difference` value in Settings-Tab (Default: 0.5 sec)
2. Sync the clock of both hosts
3. Set the current time of localhost to `odroid`-time using the button beside `username`-combobox in the yellow control panel.

regards,
atiderko

Georacer, 2016-04-05:
Hello @atiderko 

Thank you for your swift reply. Alas, my follow-up was not so swift.

On 1): I can find the code for said settings tab (https://github.com/fkie/multimaster_fkie/blob/f3af64bd1b733e64615b5960316975e9cb8567f4/node_manager_fkie/src/node_manager_fkie/settings_widget.py) but for the love of me I cannot find how to open this tab from node_manager. A screenshot with the location of the button would be helpful.
I take it you are not referring to the heartbeat rate.

On 2): Do you mean that I should eg connect both PCs to the internet and let them auto-adjust their clocks, or something more involved?

On 3): If I start node_manager on the laptop, I can only command a time-set (to laptop-time) on the Odroid. That is initially rejected due to lack of su rights, but the error message explain what to do about that.

Additionally, the same configuration today seems to work without errors (no weird colours!) Perhaps I was having an intermittent problem.

Other than that, could you tell me the meaning of the icon and its colour, which is on the top left, between the sync toggle button and the hostname and looks like a plug?
Thanks.

P.S. If you think we're done here, you can go ahead and close this ticket.

atiderko, 2016-04-06:
Hello @Georacer,

1) I see your settings-tab is not displayed. Normally it is displayed beside on tabs with launchfiles and description:
[image]
Did you have error outputs in the console where  node_manager was started? Try to delete $HOME/.ros/node_manager/settings.ini, perhaps it helps.

2) Yes, or you setup your own NTP-server ;-)

3) It was a designe decision: time change option is a temporary fix, so you have to change only the localhost.

I let this ticket open until we find the settings tab ;-)

regards

Georacer, 2016-04-06:
Yup! That did it. I can see it now, thanks!

The multimaster stack seems **really** useful for what I want to do.
Are there other documentation resources apart from 
http://fkie.github.io/multimaster_fkie/
and
http://www.iri.upc.edu/files/scidoc/1607-Multi-master-ROS-systems.pdf
?

closed 2016-04-06 by Georacer

atiderko, 2016-04-06:
there are also a chapter in http://www.springer.com/us/book/9783319260525