fkie/multimaster_fkie #37: Local and global topics/services

aswinhope, 2016-06-21:
I'm new to the multimaster concept. 

I have a situation where I have multiple robots running local ros networks and would like to facilitate communication between these networks using ROS. Since each local network uses the same set of topics, the common topics have to be isolated and the topics/services that need for inter robot communication have to be enabled. Is this possible? Or is using namespace the only way to solve this?

atiderko, 2016-06-24:
Hi,

the best way is to use namespaces, especially if you use tf. In simple cases you can also try to put topic/services into ignore lists (`~ignore_nodes`, `~ignore_topics`, `~ignore_services` or `do_not_sync`- new uncommented parameter for `master_discovery` and `master_sync`) to skip these while sync.

The multimaster_fkie package does not support automatic namespace creation.

aswinhope, 2016-06-24:
Thanks! ignore_nodes is useful

closed 2016-06-24 by aswinhope