fkie/multimaster_fkie #39: node_manager on Ubuntu 16.04 and ROS Kinetic

rikba, 2016-07-07:
Hi,
I checked out the kinetic-devel branch and built it successfully within my catkin workspace. 

On startup the node_manager crashes. Maybe due to the change from qt4 to qt5 in ROS kinetic? Here is the error message I get:

> Traceback (most recent call last):
  File "/home/rikba/catkin_ws/devel/bin/node_manager", line 6, in <module>
    exec(fh.read())
  File "<string>", line 4, in <module>
  File "/home/rikba/catkin_ws/devel/lib/python2.7/dist-packages/node_manager_fkie/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 46, in <module>
  File "/home/rikba/catkin_ws/src/node_manager/node_manager_fkie/src/node_manager_fkie/master_view_proxy.py", line 54, in <module>
    from node_manager_fkie.detailed_msg_box import WarningMessageBox, DetailedError
  File "/home/rikba/catkin_ws/src/node_manager/node_manager_fkie/src/node_manager_fkie/detailed_msg_box.py", line 48, in <module>
    class WarningMessageBox(QtGui.QMessageBox):
AttributeError: 'module' object has no attribute 'QMessageBox'

Is there any update to be expected soon?

Thanks.

atiderko, 2016-07-08:
Hi,

I will take a look next week...

atiderko, 2016-07-11:
I fixed the include errors for qt5. I hope, I found all compatibility issues..

rikba, 2016-07-18:
Thanks. Seems to be working now.

closed 2016-07-18 by atiderko