fkie/multimaster_fkie #43: "cannot import name QApplication" on ROS kinetic & Ubuntu 16.04

Sr4l, 2016-09-05:
I can't run node_manager on Ubuntu 16.04 with ROS kinetic. I'm getting:
```
kistner@mrt-pc123:~$ rosrun node_manager_fkie node_manager
cannot import name QApplication
```

I have installed all PySide1 and all PyQt5 packages, and due too some dependencys some PySide2 and PyQt4 packages.

In a normal shell this works fine:
```
kistner@mrt-pc123:~$ python
Python 2.7.12 (default, Jul  1 2016, 15:12:24) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from python_qt_binding.QtWidgets import QApplication
```
also all of these imports work:
```from PyQt4.QtGui import QApplication``` 
```from PySide.QtGui import QApplication```
```from PyQt5.QtWidgets import QApplication```
```from PySide2.QtWidgets import QApplication```

python_qt_binding uses PyQt5 bindings maybe this is a problem?

atiderko, 2016-09-05:
Thank you for reporting!

I fixed the copy&paste error. Please, try again.

Sr4l, 2016-09-05:
Thank you for the fast response. No I understand the Qt4 <-> Qt5 problem and the try/except's.

There are two more imports to fix. 

`node_manager_fkie/src/node_manager_fkie/editor/line_edit.py` - QIcon is not part of QtWidgets
`node_manager_fkie/src/node_manager_fkie/editor/text_search_frame.py` - copy&past error

See Pull Request #44

closed 2016-09-05 by Sr4l