fkie/multimaster_fkie #48: Resync duration check?

deng02, 2016-09-14:
Was wondering if there might be plans to add a duration check in the resync feature?  So preivously this feature was causing problems for us when the masters were bouncing online/offline very quickly.  That's since been fixed (thank you!) and our tests have stabalized but as additional protection, we would really like to have a configurable period that controls how long the offline-online period is before the resync kicks in.  Especially since this is supposed to address long connection timeout periods.

atiderko, 2016-09-20:
it is a good proposal!

I implemented a new parameter `~resync_on_reconnect_timeout` (default: 0). If it works like you desire I will add a corresponding tag.

deng02, 2016-10-04:
This looks like exactly like what we need. Thanks!

atiderko, 2016-10-14:
it is now in release [0.6.0](https://github.com/fkie/multimaster_fkie/releases/tag/0.6.0)

regards

closed 2016-10-14 by atiderko