fkie/multimaster_fkie #53: Not getting topics in it's own master

henriqueribeiro, 2016-11-11:
Hey guys,

I have two computers (A and B) connected to each other. Each one is running it's own roscore and multimaster_discovery and multimaster_sync. In normal operation everything is going well, all the topics appear on both sides.

However I'm getting a problem when the connection breaks. Imagine, I have a topic being published on computer A and I can see it on computer A and B. Now, if I disconnect the ethernet cable on computer B, obviously I cannot get the topic on computer B, but I also can't get it on computer A!!

I don't know exactly how multimaster works under the hood, but I think it doesn't make sense that you cannot get a topic that is being published on it's own roscore.

Hope that you can help me.
Thanks in advance!

atiderko, 2016-11-11:
Hi,

what happen if you do not start master_discover and master_sync? 
We have also observed this behavior. One case is if the ROS_MASTER_URI referenced to a hostname that is different to the entry in /etc/hostname
There are also other cases possible (e.g. NetworkManager shutdown all interfaces if no cable is plugged in). But it is not caused by multimaster_fkie. The connection between two topics is estblished and handled by ROS itself. 

If it should work after only master_discovery and master_sync, please send me further infos about network configuration. (routing table, hostname resolution...)

Thanks and regards!

henriqueribeiro, 2016-11-13:
> There are also other cases possible (e.g. NetworkManager shutdown all interfaces if no cable is plugged in)

This was exactly the problem. I removed the NetworkManager, configured /etc/network/interfaces and it worked just fine. NetworkManager disable the interface if there is no cable connected. Defining the IP on the network interfaces it keeps the IP, so communications with rosmaster works perfectly fine!

Thanks a lot @atiderko

closed 2016-11-13 by henriqueribeiro