fkie/multimaster_fkie #55: "Hub clients" can discover Hub but not vice versa

deng02, 2016-12-12:
Hi,

So we're attempting to setup a hub toplogy using the new listen_mcast and send_mcast settings.  Our hub clients have listen_mcast set to True and send_mcast set to False.  The hub itself has listen_mcast and send_mcast set to True.

When we run in this type of configuration our clients can see the hub just fine (and not each other which is great!) but unfortunately the hub can't see the clients since the client never sends out anything as part of its heartbeat.  We actually need them to see each other as there is topics and services that must be synchronized between the two.

A simple change that would allow a client to see a hub and vice versa would be to modify the send_heartbeat() to send the master discovery current state to each address in self._address if listen_mcast is True but send_mcast is False:
```
# New method for sending out current state just to addresses that have been detected
 def _send_current_state2addresses(self):
        try:
            if not self._send_mcast and self._listen_mcast:
                msg = self._create_current_state_msg()
                if msg is not None:
                    for addr in self._address:
                      rospy.logdebug('Send current state to address %s' % addr)
                      self.socket.send2addr(msg, addr)
        except Exception as e:
            rospy.logwarn('Send current state to addresses failed: %s\n' % e)
            self._init_socket()

 def send_heartbeat(self):
      with self.__lock:
            try:
                self._timer_heartbeat.cancel()
            except:
                pass
            invalid_uri = (self.master_monitor.getMasteruri() is None)
            invalid_state = (self.master_monitor.getCurrentState() is None)
            if not (invalid_uri or invalid_state or rospy.is_shutdown() or self.do_finish):
              if self._send_mcast:
                self._send_current_state2group()
             elif self._listen_mcast:
               # So in the case where send_mcast is disabled but not listen_mcast, we allow sending
               # of our state to known addresses (the hubs basically)
               self._send_current_state2addresses()
```

I'm not sure what you think of this.

Thanks.

atiderko, 2016-12-13:
Hi,

you are right, there is a bug. But the new fix leads also to a deadlock. I will try to fix it.

regards

atiderko, 2016-12-14:
Hi,

I fixed this issue. Can you try it out, please.

regards

deng02, 2016-12-14:
The issue appears to have been fixed based on my initial testing of the latest master branch. Thanks!  I consider this solved.  If possible can we get the master branch tagged or have a merge into indigo-devel so we can start testing the fixes in our larger test bed?

atiderko, 2016-12-16:
I merged the changed into indigo-devel.

deng02, 2017-01-05:
@atiderko  Happy New Year!  This bug is fixed and can be closed.  Would it be possible to get a release?

atiderko, 2017-01-06:
Happy New Year!

there are some other small bugs in node manager. I will fix it and then create a release.

closed 2017-01-06 by atiderko

atiderko, 2017-01-10:
@deng02: I created a release [0.7.0](https://github.com/fkie/multimaster_fkie/releases/tag/0.7.0)