fkie/multimaster_fkie #57: multicast not enabled

halpme96, 2017-01-11:
Hi , i am currently trying to enable a multicast on odroid UX4 but the multicast is disabled 

$ rosrun master_discovery_fkie master_discovery _send_mcast:=False
[INFO] [WallTime: 1484135078.357509] ROS Master URI: http://flytpod:11311
[WARN] [WallTime: 1484135078.384945] This master_discovery is invisible because it send no heart beat messages! Set ~send_mcast to true or add hosts to ~robot_hosts.
[WARN] [WallTime: 1484135078.385453] Multicast is disabled. Use ~active_request_after(60.00) ot set ~heartbeat_hz to new value: 0.0167
[INFO] [WallTime: 1484135078.385826] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484135078.386175] Heart beat [Hz]: 0.0166666666667
[INFO] [WallTime: 1484135078.386513] Active request after [sec]: 60
[INFO] [WallTime: 1484135078.386849] Remove after [sec]: 300
[INFO] [WallTime: 1484135078.387184] Robot hosts: []
[INFO] [WallTime: 1484135078.387519] Approx. mininum avg. network load: 0.00 bytes/s
[WARN] [WallTime: 1484135078.420415] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...
[INFO] [WallTime: 1484135079.422955] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484135079.433599] Subscribe to parameter `/roslaunch/uris`
[WARN] [WallTime: 1484135079.497026] Multicast disabled!
[INFO] [WallTime: 1484135079.500755] + Bind to unicast socket @(172.22.23.25:11511)
[INFO] [WallTime: 1484135079.721993] Detected master discovery: http://172.22.23.25:11611
[INFO] [WallTime: 1484135079.846462] Added master with ROS_MASTER_URI=http://flytpod:11311/
shutdown request: new node registered with same name
[INFO] [WallTime: 1484135330.886013] Unsubscribe from parameter `/roslaunch/uris`
[INFO] [WallTime: 1484135330.913508] shutdown own RPC server

$  rosrun master_discovery_fkie master_discovery _send_mcast:=False

Error while set the log level: [Errno 111] Connection refused
->INFO level will be used!
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/master_discovery_fkie/master_discovery", line 4, in <module>
    master_discovery_fkie.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/master_discovery_fkie/__init__.py", line 97, in main
    rospy.init_node(PROCESS_NAME, log_level=log_level)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 504, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1013, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 864, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 826, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 807, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err

halpme96, 2017-01-11:
when i tried ping 224.0.0.1 , i can't get their ros network ping

atiderko, 2017-01-12:
Hi, while start of the ROS node you set the parameter to `send_mcast` to `False`. This disables the multicast support!
In new version of multimaster this parameter is split into `send_mcast` and `listen_mcast`. The multimaster support is disabled only if both parameter are set to `False`.

The error in the second call occurs because the roscore was not running.

halpme96, 2017-01-13:
Hi, currently try to connect to odroid ux4 together using ros multi master . However , they doesn't discover each other but they can ping eachother . I don't know what goes wrong, it would be nice if you can give me some advice or help ?

Ip addresses for flytpod: 172.22.89.1 
Flytpod : 
flytpod@flytpod:~$ rosrun master_discovery_fkie master_discovery _robot_hosts:=[flytpod]
[INFO] [WallTime: 1524373808.152873] ROS Master URI: http://flytpod:11311
[INFO] [WallTime: 1524373808.192861] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1524373808.194296] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1524373808.195491] Active request after [sec]: 60
[INFO] [WallTime: 1524373808.196670] Remove after [sec]: 300
[INFO] [WallTime: 1524373808.198015] Robot hosts: ['flytpod']
[INFO] [WallTime: 1524373808.199080] Approx. mininum avg. network load: 2.72 bytes/s
[WARN] [WallTime: 1524373808.242644] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...
[INFO] [WallTime: 1524373809.247668] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1524373809.263190] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1524373809.337407] Listen for multicast at ('224.0.0.1', 11511)
[INFO] [WallTime: 1524373810.139359] Detected master discovery: http://172.22.89.1:11611
[INFO] [WallTime: 1524373810.148170] Detected master discovery: http://127.0.0.1:11611
[INFO] [WallTime: 1524373810.275012] Added master with ROS_MASTER_URI=http://flytpod:11311/
[INFO] [WallTime: 1524373810.300261] Added master with ROS_MASTER_URI=http://flytpod:11311/

rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
[INFO] [WallTime: 1482382830.309873] ROS Master URI: http://flytpod:11311
[INFO] [WallTime: 1482382830.343198] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1482382830.343699] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1482382830.344043] Active request after [sec]: 60
[INFO] [WallTime: 1482382830.344373] Remove after [sec]: 300
[INFO] [WallTime: 1482382830.344704] Robot hosts: []
[INFO] [WallTime: 1482382830.345037] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1482382830.417038] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1482382830.418214] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1482382830.437392] Listen for multicast at ('224.0.0.1', 11511)
[INFO] [WallTime: 1482382830.979373] Detected master discovery: http://172.22.89.1:11611
[INFO] [WallTime: 1482382831.117710] Added master with ROS_MASTER_URI=http://flytpod:11311/
 [INFO] [WallTime: 1516597490.004845] Remove master discovery: http://172.22.89.1:11611
[INFO] [WallTime: 1516597490.029442] Detected master discovery: http://172.22.89.1:11611
[INFO] [WallTime: 1516597490.170787] Added master with ROS_MASTER_URI=http://flytpod:11311/

127.0.0.1	localhost
127.0.1.1	flytpod

172.22.29.136	flytpod2

172.22.89.1	flytpod

::1	ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters   
Ip addresses for flytpod2 : 172.22.29.136
flytpod2:
$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
[INFO] [WallTime: 1484007485.946552] ROS Master URI: http://flytpod2:11311
[INFO] [WallTime: 1484007486.016856] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484007486.017337] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1484007486.017682] Active request after [sec]: 60
[INFO] [WallTime: 1484007486.018013] Remove after [sec]: 300
[INFO] [WallTime: 1484007486.018345] Robot hosts: []
[INFO] [WallTime: 1484007486.018676] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1484007486.050277] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484007486.051474] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1484007486.061122] Listen for multicast at ('224.0.0.1', 11511)

[INFO] [WallTime: 1484007486.280585] Detected master discovery: http://172.22.29.136:11611
[INFO] [WallTime: 1484007486.403779] Added master with ROS_MASTER_URI=http://flytpod2:11311/
shutdown request: new node registered with same name
[INFO] [WallTime: 1484007631.073526] Unsubscribe from parameter `/roslaunch/uris`
[INFO] [WallTime: 1484007631.085512] shutdown own RPC server
[INFO] [WallTime: 1484007631.418428] Stop receiving multicast at ('224.0.0.1', 11511)

Ip addresses for flytpod2 : 172.22.29.136
$ rosrun master_discovery_fkie master_discovery _robot_hosts:=[flytpod2]
[INFO] [WallTime: 1484007631.086680] ROS Master URI: http://flytpod2:11311
[INFO] [WallTime: 1484007631.120316] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484007631.120799] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1484007631.121144] Active request after [sec]: 60
[INFO] [WallTime: 1484007631.121481] Remove after [sec]: 300
[INFO] [WallTime: 1484007631.121816] Robot hosts: ['flytpod2']
[INFO] [WallTime: 1484007631.122149] Approx. mininum avg. network load: 2.72 bytes/s
[WARN] [WallTime: 1484007631.155251] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...
[INFO] [WallTime: 1484007632.157573] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484007632.165283] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1484007632.204563] Listen for multicast at ('224.0.0.1', 11511)
[INFO] [WallTime: 1484007632.498089] Detected master discovery: http://172.22.29.136:11611
[INFO] [WallTime: 1484007632.637996] Added master with ROS_MASTER_URI=http://flytpod2:11311/

127.0.0.1    localhost
127.0.1.1    flytpod2

172.22.89.1    flytpod

172.22.29.136    flytpod2

::1    ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters

atiderko, 2017-01-13:
Please change the `robot_hosts` value to
on flypod: rosrun master_discovery_fkie master_discovery _robot_hosts:=[flytpod2]
on flypod2: rosrun master_discovery_fkie master_discovery _robot_hosts:=[flytpod]

This should work. Then we can try to solve the problem why the multicast not work.
1. are flypod and flypod2 in the same network?
2. how looks the routing table `route -n`?

halpme96, 2017-01-13:
it works already . thank you.HAHA.  i now need to connect 3 odroid within the same network , is it possible to use the same method as i used previously ?

atiderko, 2017-01-13:
yes, you can specify in `robot_hosts` all other hosts, e.g. _robot_hosts:=[flytpod, flytpod2]

halpme96, 2017-01-13:
why all their two master is not online ? although they can discover eachother but is it actually working fine ? (trying with 3 ros network) 
it will auto set host to offline is there a way to set it online longer ? 
the ros multimaster will auto shut off after 10mins , is there a way to on them forever? 
masters: 
  - 
    name: flytpod2
    uri: http://flytpod2:11311/
    timestamp: 1484021419.36
    timestamp_local: 1484021419.36
    online: True
    discoverer_name: /master_discovery
    monitoruri: http://172.22.29.136:11611
  - 
    name: flytpod
    uri: http://flytpod:11311/
    timestamp: 1524386163.86
    timestamp_local: 0.0
    online: False
    discoverer_name: /master_discovery
    monitoruri: http://172.22.89.1:11611
  - 
    name: wq
    uri: http://wq:11311/
    timestamp: 1484296264.19
    timestamp_local: 0.0
    online: False
    discoverer_name: /master_discovery
    monitoruri: http://172.22.69.139:11611

flytpod2:
route -n 
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         172.22.24.1     0.0.0.0         UG    0      0        0 wlan5
172.22.24.0     0.0.0.0         255.255.248.0   U     9      0        0 wlan5

flytpod:
route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         172.22.88.1     0.0.0.0         UG    0      0        0 wlan4
172.22.88.0     0.0.0.0         255.255.248.0   U     9      0        0 wlan4

atiderko, 2017-01-13:
1. why all their two master is not online: 
I think it is a bug in the used multimaster version. You can try to update or set the `_heartbeat_hz:=0.5`

2. it will auto set host to offline is there a way to set it online longer: 
it sets the host to offline, because no heartbeats are received from this host. Try the solution in 1.. 

3. the ros multimaster will auto shut off after 10mins: 
There is no mechanism in multimaster for auto shut off. If you mean, that the offline host are removed from discovered list, you can change this by `remove_after` (seconds after is was set to offline) parameter.

Are the hosts in different Wifi Networks? Perhaps is it the reason, why the multicast solution does not work.

halpme96, 2017-01-16:
hi ,  i still got problems with the host offline and there is one more question i want to ask you , what can ros multimaster do ? is it possible to trigger their ros network to do a certain cpp file ? 
rosrun master_discovery_fkie master_discovery _heartbeat_hz:=0.5
[INFO] [WallTime: 1484526061.385719] ROS Master URI: http://flytpod:11311
[INFO] [WallTime: 1484526061.431633] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484526061.432197] Heart beat [Hz]: 0.5
[INFO] [WallTime: 1484526061.432609] Active request after [sec]: 60
[INFO] [WallTime: 1484526061.433002] Remove after [sec]: 9999
[INFO] [WallTime: 1484526061.433401] Robot hosts: ['flytpod2']
[INFO] [WallTime: 1484526061.433796] Approx. mininum avg. network load: 68.00 bytes/s
[INFO] [WallTime: 1484526061.471033] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484526061.472246] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1484526061.482970] Listen for multicast at ('224.0.0.1', 11511)
[INFO] [WallTime: 1484526062.313836] Detected master discovery: http://172.22.23.39:11611
[INFO] [WallTime: 1484526062.327334] Detected master discovery: http://172.22.27.201:11611
[INFO] [WallTime: 1484526062.473971] Added master with ROS_MASTER_URI=http://flytpod:11311/
[INFO] [WallTime: 1484526063.188435] Added master with ROS_MASTER_URI=http://flytpod2:11311/
[INFO] [WallTime: 1484526140.747459] Set host to offline: flytpod2

atiderko, 2017-01-16:
Hi, did you run master_discovery also on `flypod2` with `_heartbeat_hz` parameter?:

The master_discovery detects by sending and receiving of hearbeats (small UDP messages) other master_discovery nodes. The heartbeats are also used to announce the changes on local roscore and to test if a detected master_discovery node is still reachable. On changes a master_dicovery publish a message to ROS topic [`/master_discovery/changes`](http://docs.ros.org/api/multimaster_msgs_fkie/html/msg/MasterState.html). You can create a new ROS node, listen to the topic and do your stuff if needed.

halpme96, 2017-01-16:
i download multimaster by typing "sudo apt-get installl ros-indigo-multimaster-fkie" 

so is it possible to trigger the files at the same time for like 3 ROS Network ? is there any tutorial to teach me how to create node or do the stuff i want ? (more in depth)

atiderko, 2017-01-16:
It would be not at the same time. Changes on local roscore are earlier. It may take several seconds until remote nodes take notice about the change.
To get the notice about a change (new Host discovered, one changed or removed) you need to subscribe to the topic `/master_discovery/changes` ([type](http://docs.ros.org/api/multimaster_msgs_fkie/html/msg/MasterState.html)), see [ROS Subscriber Tutorial](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber(c%2B%2B)#roscpp_tutorials.2BAC8-Tutorials.2BAC8-WritingPublisherSubscriber.Writing_the_Subscriber_Node). Use `multimaster_msgs_fkie/MasterState` for callback type. In the `callback` method you can use the message to determine which kind of changes it is and propertely react.

halpme96, 2017-01-16:
Actually i am working on 3 drone with odroid on each , i was thinking whether is it possible by using ros multi-master . The ground station will be my local host and 3 other drone(subscriber) will be taking off at the same time(not sure whether is it possible) after they received the rosnode/rostopic from the ground station(publisher) ?

atiderko, 2017-01-16:
I think it make more sense to implement this feature without the multimaster_fkie. Create a subscriber on each drone, which subscribe to the same topic. The publisher is then on the ground station and publish a go command.
You can then use master_discovery and master_sync to run on each drone their own `roscore` and synchronize all topics.

halpme96, 2017-01-20:
hi , i still got 1 problem , it will set my their computer to offline host and remove their master discovery then detected and added the master . The cycle will repeat .

atiderko, 2017-01-20:
hi, can you use the multimaster_fkie from github?
or try with [ROS ShadowRepository](http://wiki.ros.org/ShadowRepository) to get the current version.

halpme96, 2017-01-20:
after i download as a zip file , how do i get it to replace my older version? what command must i run?

atiderko, 2017-01-20:
`unzip file.zip -d ros_workspace/src/`
or
you got to your ros_workspace/src/ and call `git clone https://github.com/fkie/multimaster_fkie.git`

halpme96, 2017-01-20:
after i download it in my catkin_ws/src , it still have problem with the set host to offline

atiderko, 2017-01-20:
did you this on all your hosts, which are involved. And did you set on **all** hosts `_robot_hosts` `_heartbeat_hz`?

halpme96, 2017-01-20:
currently working on two ros computers i had set the _robot_hosts to eachother and heartbeat i set to 0.5 hz

atiderko, 2017-01-20:
please, run twice
`rosrun master_discovery_fkie master_discovery _robot_hosts:=[flytpod] _heartbeat_hz:=0.5 _log_level:=DEBUG`

and post the output

halpme96, 2017-01-20:
rosrun master_discovery_fkie master_discovery _robot_hosts:=[ux302lg] _heartbeat_hz:=0.5 _log_level:=DEBUG
[DEBUG] [WallTime: 1484903800.813327] init_node, name[/master_discovery], pid[3424]
[DEBUG] [WallTime: 1484903800.816013] binding to 0.0.0.0 0
[DEBUG] [WallTime: 1484903800.817788] bound to 0.0.0.0 50954
[DEBUG] [WallTime: 1484903800.822284] ... service URL is rosrpc://flytpod:50954
[DEBUG] [WallTime: 1484903800.824245] [/master_discovery/get_loggers]: new Service instance
[DEBUG] [WallTime: 1484903800.844588] ... service URL is rosrpc://flytpod:50954
[DEBUG] [WallTime: 1484903800.847324] [/master_discovery/set_logger_level]: new Service instance
[INFO] [WallTime: 1484903800.954679] ROS Master URI: http://flytpod:11311
[INFO] [WallTime: 1484903800.987350] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484903800.987917] Heart beat [Hz]: 0.5
[INFO] [WallTime: 1484903800.988279] Active request after [sec]: 60
[INFO] [WallTime: 1484903800.988611] Remove after [sec]: 300
[INFO] [WallTime: 1484903800.988943] Robot hosts: ['ux302lg']
[INFO] [WallTime: 1484903800.989280] Approx. mininum avg. network load: 68.00 bytes/s
[INFO] [WallTime: 1484903801.024151] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484903801.025392] Subscribe to parameter `/roslaunch/uris`
[DEBUG] [WallTime: 1484903801.035643] mgroup: 226.0.0.0
[DEBUG] [WallTime: 1484903801.036206] interface : None
[DEBUG] [WallTime: 1484903801.036574] inet: (2, 1, 6, '', ('226.0.0.0', 0))
[INFO] [WallTime: 1484903801.036977] Listen for multicast at ('226.0.0.0', 11511)
[DEBUG] [WallTime: 1484903801.058404] node[/master_discovery, http://flytpod:41657/] entering spin(), pid[3424]
[DEBUG] [WallTime: 1484903801.872334] Received a multicast request for a state update from 172.22.58.149
[DEBUG] [WallTime: 1484903801.881048] Send current state to group 226.0.0.0:11511
[INFO] [WallTime: 1484903801.892432] Detected master discovery: http://172.22.58.149:11611
[INFO] [WallTime: 1484903801.905176] Detected master discovery: http://172.22.94.179:11611
[DEBUG] [WallTime: 1484903802.004648] Get additional connection info from http://172.22.58.149:11611
[DEBUG] [WallTime: 1484903802.017317] Get additional connection info from http://172.22.94.179:11611
[DEBUG] [WallTime: 1484903802.026698] Got [1484903801.156409979, http://flytpod:11311/, flytpod, /master_discovery] from http://flytpod:11611/
[INFO] [WallTime: 1484903802.029946] Added master with ROS_MASTER_URI=http://flytpod:11311/
[DEBUG] [WallTime: 1484903802.033761] Got [1484903816.642431974, http://UX302LG:11311/, ux302lg, /master_discovery] from http://UX302LG:11611/
[DEBUG] [WallTime: 1484903802.038018] ... service URL is rosrpc://flytpod:50954
[INFO] [WallTime: 1484903802.038404] Added master with ROS_MASTER_URI=http://UX302LG:11311/
[DEBUG] [WallTime: 1484903802.041589] [/master_discovery/list_masters]: new Service instance
[DEBUG] [WallTime: 1484903802.050058] ... service URL is rosrpc://flytpod:50954
[DEBUG] [WallTime: 1484903802.051886] [/master_discovery/refresh]: new Service instance
[DEBUG] [WallTime: 1484903802.060744] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903802.062958] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903802.065354] Set timer to send heartbeat in 2.00 sec
[DEBUG] [WallTime: 1484903803.840002] Received a multicast request for a state update from 172.22.58.149
[DEBUG] [WallTime: 1484903803.849311] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903804.069846] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903804.083386] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903804.095915] Set timer to send heartbeat in 2.00 sec
[DEBUG] [WallTime: 1484903805.961984] Received a multicast request for a state update from 172.22.58.149
[DEBUG] [WallTime: 1484903805.971812] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903806.107685] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903806.117117] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903806.127359] Set timer to send heartbeat in 2.00 sec
[DEBUG] [WallTime: 1484903807.841930] Received a multicast request for a state update from 172.22.58.149
[DEBUG] [WallTime: 1484903807.885313] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903808.140139] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903808.151997] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903808.164556] Set timer to send heartbeat in 2.00 sec
[DEBUG] [WallTime: 1484903809.930868] Received a multicast request for a state update from 172.22.58.149
[DEBUG] [WallTime: 1484903809.940173] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903810.176496] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903810.188332] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903810.201163] Set timer to send heartbeat in 2.00 sec
[DEBUG] [WallTime: 1484903812.019998] Received a multicast request for a state update from 172.22.58.149
[DEBUG] [WallTime: 1484903812.028853] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903812.219563] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903812.230650] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903812.241169] Set timer to send heartbeat in 2.00 sec
[... cut ...]
[DEBUG] [WallTime: 1484903900.075867] Send a unicast request for update to 172.22.58.149
[DEBUG] [WallTime: 1484903900.078081] Unanswered requests [5] for http://UX302LG:11311/: [1484903891.903651, 1484903893.938111, 1484903895.974173, 1484903898.035501, 1484903900.07766]
[DEBUG] [WallTime: 1484903901.930835] Send current state to group 226.0.0.0:11511
[DEBUG] [WallTime: 1484903901.943047] Send a unicast request for update to ux302lg
[DEBUG] [WallTime: 1484903901.955816] Set timer to send heartbeat in 2.00 sec
[INFO] [WallTime: 1484903901.967964] Set host to offline: ux302lg

halpme96, 2017-01-20:
The first time i download my multimaster fkie through sudo-apt-get install ros-indigo-multimaster-fkie

atiderko, 2017-01-20:
this is the output of the multimaster version 0.6.2. The new version from github or from [ROS ShadowRepository](http://wiki.ros.org/ShadowRepository) has a fix for networks with multicast problems. Please try it out.
You has already downloaded a zip file or maked a git clone of new version. To be sure not to use the old installed version of mutlimaster_fkie please uninstall it with `sudo apt purge ros-indigo-multimaster-fkie`

halpme96, 2017-01-20:
after downloading zip file is there any configure need to be make ?

atiderko, 2017-01-20:
you have to put the content of the zip file into your ROS_WORKSPACE/src and `caktin build`. You have also call `source ROS_WORKSPACE/devel.setup.bash`. Then you can start in the same terminal the master_discovery

halpme96, 2017-01-20:
$ cd catkin_ws 
nishinihon201@UX302LG:~/catkin_ws$ ls
build  devel  install  src
nishinihon201@UX302LG:~/catkin_ws$ catkin_make 
Base path: /home/nishinihon201/catkin_ws
Source space: /home/nishinihon201/catkin_ws/src
Build space: /home/nishinihon201/catkin_ws/build
Devel space: /home/nishinihon201/catkin_ws/devel
Install space: /home/nishinihon201/catkin_ws/install
####
#### Running command: "cmake /home/nishinihon201/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/nishinihon201/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/nishinihon201/catkin_ws/install -G Unix Makefiles" in "/home/nishinihon201/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/nishinihon201/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/nishinihon201/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/nishinihon201/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nishinihon201/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - multimaster_fkie (metapackage)
-- ~~  - multimaster_msgs_fkie
-- ~~  - default_cfg_fkie
-- ~~  - master_discovery_fkie
-- ~~  - beginner_tutorials
-- ~~  - master_sync_fkie
-- ~~  - node_manager_fkie
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'multimaster_fkie'
-- ==> add_subdirectory(multimaster_fkie-0.7.0/multimaster_fkie)
-- +++ processing catkin package: 'multimaster_msgs_fkie'
-- ==> add_subdirectory(multimaster_fkie-0.7.0/multimaster_msgs_fkie)
-- Using these message generators: gencpp;genlisp;genpy
-- multimaster_msgs_fkie: 8 messages, 6 services
-- +++ processing catkin package: 'default_cfg_fkie'
-- ==> add_subdirectory(multimaster_fkie-0.7.0/default_cfg_fkie)
-- +++ processing catkin package: 'master_discovery_fkie'
-- ==> add_subdirectory(multimaster_fkie-0.7.0/master_discovery_fkie)
-- +++ processing catkin package: 'beginner_tutorials'
-- ==> add_subdirectory(beginner_tutorials)
-- +++ processing catkin package: 'master_sync_fkie'
-- ==> add_subdirectory(multimaster_fkie-0.7.0/master_sync_fkie)
-- +++ processing catkin package: 'node_manager_fkie'
-- ==> add_subdirectory(multimaster_fkie-0.7.0/node_manager_fkie)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nishinihon201/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/nishinihon201/catkin_ws/build"
####
[  0%] [  0%] [  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_ListNodes
Built target _multimaster_msgs_fkie_generate_messages_check_deps_Task
Built target _multimaster_msgs_fkie_generate_messages_check_deps_ROSMaster
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_LinkState
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_SyncTopicInfo
[  0%] [  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_Capability
Built target _multimaster_msgs_fkie_generate_messages_check_deps_MasterState
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_LinkStatesStamped
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_SyncServiceInfo
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_DiscoverMasters
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_SyncMasterInfo
[  0%] Built target _multimaster_msgs_fkie_generate_messages_check_deps_ListDescription
[... 36 lines cut ...]
[ 75%] [ 77%] Generating Python code from SRV multimaster_msgs_fkie/ListDescription
Generating Lisp code from multimaster_msgs_fkie/ListDescription.srv
[ 79%] [ 81%] Generating C++ code from multimaster_msgs_fkie/DiscoverMasters.srv
Generating Python code from SRV multimaster_msgs_fkie/Task
[ 84%] Generating Lisp code from multimaster_msgs_fkie/Task.srv
[ 86%] Generating C++ code from multimaster_msgs_fkie/ListNodes.srv
[ 86%] [ 88%] Built target multimaster_msgs_fkie_generate_messages_lisp
[ 90%] Generating Python msg __init__.py for multimaster_msgs_fkie
Generating Python srv __init__.py for multimaster_msgs_fkie
[ 93%] [ 93%] Built target multimaster_msgs_fkie_generate_messages_py
Generating C++ code from multimaster_msgs_fkie/GetSyncInfo.srv
[ 95%] Generating C++ code from multimaster_msgs_fkie/LoadLaunch.srv
[ 97%] Generating C++ code from multimaster_msgs_fkie/ListDescription.srv
[100%] Generating C++ code from multimaster_msgs_fkie/Task.srv
[100%] Built target multimaster_msgs_fkie_generate_messages_cpp
[100%] Built target multimaster_msgs_fkie_generate_messages
nishinihon201@UX302LG:~/catkin_ws$ source /home/nishinion201/catkin_ws/devel.setup.bash
bash: /home/nishinion201/catkin_ws/devel.setup.bash: No such file or directory
nishinihon201@UX302LG:~/catkin_ws$ source /home/nishinion201/catkin_ws/devel/setup.bash
bash: /home/nishinion201/catkin_ws/devel/setup.bash: No such file or directory
nishinihon201@UX302LG:~/catkin_ws$ source /home/nishinihon201/catkin_ws/devel/setup.bash
nishinihon201@UX302LG:~/catkin_ws$ cd ~
nishinihon201@UX302LG:~$ rosrun master_sync_fkie master_sync
rosrun: command not found
can you help ?

halpme96, 2017-01-20:
it keeps saying command not found

atiderko, 2017-01-20:
try with `source /opt/ros/indigo/setup.bash`

halpme96, 2017-01-20:
don't seem to work

atiderko, 2017-01-20:
after `source /opt/ros/indigo/setup.bash` you have to call `source /home/nishinion201/catkin_ws/devel/setup.bash`

halpme96, 2017-01-20:
nishinihon201@UX302LG:~/catkin_ws$ source /opt/ros/indigo/setup.bash
nishinihon201@UX302LG:~/catkin_ws$ source /home/nishinihon201/catkin_ws/devel/setup.bash
nishinihon201@UX302LG:~/catkin_ws$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
rosrun: command not found

halpme96, 2017-01-20:
it works on a odroid but don't work on a laptop running on ubuntu

atiderko, 2017-01-20:
it seems that is something wrong with your ros installation on ubuntu. The system does not find `rosrun`. I do not know what can help; reinstall, reboot...

halpme96, 2017-01-24:
Traceback (most recent call last):
  File "/home/flytpod/catkin_ws/src/multimaster_fkie-0.7.0/master_discovery_fkie/nodes/master_discovery", line 4, in <module>
    master_discovery_fkie.main()
  File "<string>", line 90, in main
  File "/home/flytpod/catkin_ws/src/multimaster_fkie-0.7.0/master_discovery_fkie/src/master_discovery_fkie/master_discovery.py", line 33, in <module>
    from hgext.convert.convcmd import recode
ImportError: No module named hgext.convert.convcmd

halpme96, 2017-01-24:
is there anything wrong here

atiderko, 2017-01-24:
you're right. I fixed it, please update.

halpme96, 2017-01-24:
currently using kinetic , is it available in multimaster ?

halpme96, 2017-01-24:
how do i update ?

atiderko, 2017-01-24:
I think it is easier you remove `multimaster_fkie-0.7.0` from `src` folder. Go to `catkin_ws/src` and call 
`git clone https://github.com/fkie/multimaster_fkie.git`

halpme96, 2017-01-24:
rosrun node_manager_fkie node_managerTraceback (most recent call last):
  File "/home/flytpod/catkin_ws/src/multimaster_fkie/node_manager_fkie/nodes/node_manager", line 4, in <module>
    node_manager_fkie.main("node_manager")
AttributeError: 'module' object has no attribute 'main'
flytpod@flytpod2:~/catkin_ws/src$ rosrun node_manager_fkie node_managerTraceback (most recent call last):
  File "/home/flytpod/catkin_ws/src/multimaster_fkie/node_manager_fkie/nodes/node_manager", line 4, in <module>
    node_manager_fkie.main("node_manager")
AttributeError: 'module' object has no attribute 'main'
is there any problem here ?

halpme96, 2017-01-24:
is there anything wrong here ? 
osrun node_manager_fkie node_managerUnable to find an X11 visual which matches EGL config 9
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
Unable to find an X11 visual which matches EGL config 9
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
[INFO] [1487868985.347665]: listen for logs on /rosout
Unable to find an X11 visual which matches EGL config 9
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
[ERROR] [1487868986.529856]: No SCREEN available! You can't launch nodes.
[INFO] [1487868986.569671]: Start RPC-XML Server at ('0.0.0.0', 22622)
[INFO] [1487868986.572813]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1487868986.863892]: listen for updates on /master_discovery/changes
[INFO] [1487868986.954047]: listen for connection statistics on /master_discovery/linkstats
[INFO] [1487868987.937665]: listen for updates on /master_discovery/changes
[INFO] [1487868988.038556]: listen for connection statistics on /master_discovery/linkstats
Unable to find an X11 visual which matches EGL config 9
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
[INFO] [1487869029.451054]: Run without config: /usr/bin/screen -c /home/flytpod/.ros/log/_master_sync.conf -L -dmS _master_sync /home/flytpod/catkin_ws/src/multimaster_fkie/master_sync_fkie/nodes/master_sync _interface_url:='.' _sync_topics_on_demand:=False _ignore_hosts:=[] _sync_hosts:=[] _ignore_nodes:=[] _sync_nodes:=[] _ignore_topics:=[] _ignore_publishers:=[] _ignore_subscribers:=[] _sync_topics:=[] _ignore_services:=[] _sync_services:=[] _sync_remote_nodes:=False __name:=master_sync
[WARN] [1487869029.564142]: Start sync on flytpod2 failed:
	OSError: [Errno 2] No such file or directory
Unable to find an X11 visual which matches EGL config 9
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
Unable to find an X11 visual which matches EGL config 9
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.

atiderko, 2017-01-24:
did you call `catkin build`?

halpme96, 2017-01-24:
[WARN] [1485160757.160260]: MasterMonitor[/flytpod/mavros]: can't get PID: [Errno 111] Connection refused
is there anything wrong?

atiderko, 2017-01-24:
Not in node_manager or master_discovery. This warning is shown, when a ROS node registered in roscore does not response. It is an indication that the node was died or not properly shut down.

closed 2017-02-05 by halpme96