fkie/multimaster_fkie #58: Unable to detect other ROS Multimaster Network

mingfeng97, 2017-01-12:
Hi, I am currently having trouble detecting the other ROS network via 224.0.0.1, even though I can ping and ssh each other. The output of my master_discovery node:

Computer 1 code : ~$ rosrun master_discovery_fkie master_discovery _mcast_group:224.0.0.1
[INFO] [WallTime: 1484214084.989779] ROS Master URI: http://172.22.58.242:11311
[INFO] [WallTime: 1484214085.000409] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484214085.000930] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1484214085.001756] Active request after [sec]: 60
[INFO] [WallTime: 1484214085.001945] Remove after [sec]: 300
[INFO] [WallTime: 1484214085.002111] Robot hosts: []
[INFO] [WallTime: 1484214085.002260] Approx. mininum avg. network load: 1.36 bytes/s
[WARN] [WallTime: 1484214085.011642] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...
[INFO] [WallTime: 1484214086.013270] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484214086.014247] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1484214086.020829] Listen for multicast at ('226.0.0.0', 11511)
[INFO] [WallTime: 1484214086.144409] Detected master discovery: http://172.22.58.242:11611
[INFO] [WallTime: 1484214086.246949] Added master with ROS_MASTER_URI=http://UX302LG:11311/

Computer 2 code: 
~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1[INFO] [WallTime: 1484214037.136613] ROS Master URI: http://tay:11311
[INFO] [WallTime: 1484214037.154114] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484214037.154360] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1484214037.154531] Active request after [sec]: 60
[INFO] [WallTime: 1484214037.154684] Remove after [sec]: 300
[INFO] [WallTime: 1484214037.154829] Robot hosts: []
[INFO] [WallTime: 1484214037.154976] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1484214037.238761] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484214037.239193] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1484214037.243465] Listen for multicast at ('224.0.0.1', 11511)
[INFO] [WallTime: 1484214037.369855] Detected master discovery: http://172.22.22.86:11611
[INFO] [WallTime: 1484214037.475341] Added master with ROS_MASTER_URI=http://tay:11311/

I checked the multicast feature and both computers code shows this: 
~$ cat /proc/sys/net/ipv4/icmp_echo_ignore_broadcasts
0

Hope you'll reply soon, thanks!

atiderko, 2017-01-12:
Hi, are the multicast routes correct? What shows `route -n`

In the meantime you can test with unicast detection:
on computer 1: `rosrun master_discovery_fkie master_discovery _robot_hosts:=[tay]`
on computer 2: `rosrun master_discovery_fkie master_discovery _robot_hosts:=[172.22.58.242]`

mingfeng97, 2017-01-13:
Hi, after testing with unicast detection on computer 1 and 2. Below are the outputs.

Computer 1:
nishinihon201@UX302LG:~$ rosrun master_discovery_fkie master_discovery _robot_hosts:=[tay]
Error while set the log level: [Errno 111] Connection refused
->INFO level will be used!
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/master_discovery_fkie/master_discovery", line 4, in <module>
    master_discovery_fkie.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/master_discovery_fkie/__init__.py", line 97, in main
    rospy.init_node(PROCESS_NAME, log_level=log_level)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 504, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1013, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 864, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 826, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 807, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused

Computer 2: 
tay@tay:~$ rosrun master_discovery_fkie master_discovery _robot_hosts:=[UX302LG][INFO] [WallTime: 1484279727.448112] [3434.522000] ROS Master URI: http://tay:11311
[INFO] [WallTime: 1484279727.482614] [3434.554000] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1484279727.483022] [3434.554000] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1484279727.483686] [3434.554000] Active request after [sec]: 60
[INFO] [WallTime: 1484279727.484094] [3434.554000] Remove after [sec]: 300
[INFO] [WallTime: 1484279727.484460] [3434.558000] Robot hosts: ['UX302LG']
[INFO] [WallTime: 1484279727.484681] [3434.558000] Approx. mininum avg. network load: 2.72 bytes/s
[INFO] [WallTime: 1484279727.576511] [3434.646000] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1484279727.577116] [3434.646000] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1484279727.588649] [3434.656000] Listen for multicast at ('226.0.0.0', 11511)
[INFO] [WallTime: 1484279728.023697] [3435.056000] Detected master discovery: http://172.22.20.150:11611
[INFO] [WallTime: 1484279728.128538] [3435.160000] Added master with ROS_MASTER_URI=http://tay:11311/

Thanks for replying!

atiderko, 2017-01-13:
Hi,

Computer 2 looks good. Multicast is enabled and I think it would also work without `robot_host` parameter. You can test it, if the system works.

Computer 1: The error occurs, because the `master_discovery` node can't connect to `roscore`. Does other ROS nodes work? What is the output of the `roscore`

regards

atiderko, 2017-05-09:
@mingfeng97 is the problem still exists

SimbaXu, 2017-05-16:
Hi, 

I have a quite similar case here. 
I have several computers on the same local network. Computer 1 can detect the other computers but it cannot be detected. All other computers can detect each other. It can only be detected by Computer 2 with unicast detection: 
rosrun master_discovery_fkie master_discovery _robot_hosts:=[Computer 2].

cat /proc/sys/net/ipv4/icmp_echo_ignore_broadcasts returns 0 for all computers.

@atiderko Could you help to troubleshoot?

Thank you!

bpinaya, 2017-05-28:
@atiderko Hi there, I am trying to set up multimaster with 2 different computers, using the docs provided here: [Multi-master ROS systems](http://www.iri.upc.edu/files/scidoc/1607-Multi-master-ROS-systems.pdf) section 3 but it's not working. Bellow, you can see the logs

`rosrun master_discovery_fkie master_discovery _robot_host:=[SRC-OCU3]
[INFO] [WallTime: 1496004910.735777] ROS Master URI: http://asusrog:11311
[INFO] [WallTime: 1496004910.744149] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1496004910.744307] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1496004910.744419] Active request after [sec]: 60
[INFO] [WallTime: 1496004910.744528] Remove after [sec]: 300
[INFO] [WallTime: 1496004910.744632] Robot hosts: []
[INFO] [WallTime: 1496004910.744734] Approx. mininum avg. network load: 1.36 bytes/s
[WARN] [WallTime: 1496004910.752471] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...
shutdown request: new node registered with same name
[INFO] [WallTime: 1496004911.753929] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1496004911.757749] Create multicast socket at ('224.0.0.1', 11511)
[INFO] [WallTime: 1496004911.959113] Close multicast socket at ('224.0.0.1', 11511)
`
when I try to run:
rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
I get:
Error while start RPC-XML server on port 11611: [Errno 98] Address already in use
Try again...
Any help is appreciated!!!!

atiderko, 2017-05-29:
Hi @bpinaya,

if you see the error `Error while start RPC-XML server on port 11611: [Errno 98] Address already in use` then a second instance of the master_discovery is currently running. This will be stopped by ROS. And it should work, like in your first log. If not, try to restart your PC and run master_dsicovery only once!

`_robot_host**s**` parameter (you have a typo in the parameter, so `[INFO] [WallTime: 1496004910.744632] Robot hosts: []` is empty) you need only if multicast does not work.

What happens if you start master_discovery without any parameter?

bpinaya, 2017-05-29:
Yes, I found out that after looking which process was running on that port already, thanks @atiderko !!!

closed 2017-10-27 by atiderko