fkie/multimaster_fkie #61: Is there a way to ignore all services or all topics?

davidbsp, 2017-04-28:
I would like to be able to ignore synchronizing all services of a given master. 
Is there a way to do this using the *ignore_services* parameter of master_sync_fkie?

atiderko, 2017-04-28:
```
ignore_services:
  - /*
```
or for an host:

```
ignore_services:
  - hostname:
    - /*
```

davidbsp, 2017-05-02:
Hi @atiderko,

```
ignore_services:
  - /*
```

works fine (as well as with ignore_topics).

However, when trying this one:

```
ignore_services:
  - hostname:
    - /*
```

I get the same error as in #60. That is:

```
[WARN] [1493728025.948580]: ERROR while update master[ing3]: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/master_sync_fkie/master_sync.py", line 169, in update_master
    self.masters[mastername] = SyncThread(mastername, masteruri, discoverer_name, monitoruri, 0.0, self.__sync_topics_on_demand)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/master_sync_fkie/sync_thread.py", line 113, in __init__
    [])
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/master_discovery_fkie/filter_interface.py", line 79, in load
    sync_topics, mastername)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/master_discovery_fkie/common.py", line 276, in create_pattern
    def_list = list(set(def_list))
TypeError: unhashable type: 'dict'
```

davidbsp, 2017-05-02:
commit [aa3b870](https://github.com/fkie/multimaster_fkie/commit/aa3b870c0e566533aec6b7076014bef31765af0e) fixed the issue. Thanks!

closed 2017-05-03 by atiderko