fkie/multimaster_fkie #62: Synchonizing multiple robots crashes capability_server

BasJ93, 2017-05-29:
I'm trying to run multiple turtlebots as a multimaster system. When I sync a single turtlebot with my control computer it works.

However, when I bring a second robot into the mix, the capability_server crashes when the system syncs.

I'm trying to only sync two topics. When not running the turtlebot packages, this appears to work. However, because the capablity_server crashes when I am running the turtlebot stack, I must conclude more information is being synced between the masters.

Can you please provide me with more information, or point me to a mistake?

Attached you can find my launchfile for the multimaster_sync.

```XML
<launch>

  <node name="master_sync" pkg="master_sync_fkie" type="master_sync" output="screen">
    <!-- Performs resync after the host was offline: unregister and register again to avoid connection losses to python topic. These does not suppot reconnection! -->
    <param name="resync_on_reconnect" value="True" />
    <!-- A configurable period that controls how long the offline-online period [sec] is before the resync kicks in.
     (Since version 0.5.9)-->
    <param name="resync_on_reconnect_timeout" value="0" />
    <!-- A filter list with patterns. If this pattern merges to a node, topic or service it will not be synchronized.
     (Since version 0.5.4)-->
    <rosparam param="do_not_sync">[]</rosparam>

    <!-- This list with names of ROS master to ignore -->
    <rosparam param="ignore_hosts">[]</rosparam>

    <!-- This list with names of ROS master to synchronize. Only this ROS master in this list will be synchronized. The ~ignore_hosts list is processed first. -->
    <rosparam param="sync_hosts">[]</rosparam>

    <!-- The list of nodes, which are to be ignored while a synchroniation. All their topics and services are ignored!
    These are defaults and cannot be removed: [/rosout, "/*master_sync* node", "remote /*master_discovery* node", "/*node_manager", /*zeroconf] -->
    <rosparam param="ignore_nodes">[]</rosparam>

    <!-- The list of nodes to synchronize. Only topics of this nodes are synchronized. The ~ignore_nodes, ~ignore_topics and ~ignore_services lists are processed first. -->
    <rosparam param="sync_nodes">[]</rosparam>

    <!-- The list of topics, which are to be ignored while a synchroniation. The ~ignore_nodes list is processed first.
    These are defaults and cannot be removed: ['/rosout', '/rosout_agg'] -->
<!--    <rosparam param="ignore_topics">["/*einstein*", "/*newton*", "/*tesla*", "/*pythagoras*", "/tf", "/tf_static", "/*move_base*", "/*camera*", "/*navigation*", "/odom", "/*mobile_base*", "/*amcl*", "/*capability_server*", "/*cmd_vel_mux*", "/*depthimage_to_laserscan*", "/*diagnostics*", "/*gateway*", "/*info*", "/*initialpose*", "/*interactions*", "/*joint_states*", "/*kobuki_safety_controller*", "/*map*", "/*map_metadata*", "/*navigation_velocity_smoother*", "/particlecloud", "/rosout", "/rosout_agg", "/scan", "/*zeroconf*"]</rosparam> -->
    <rosparam param="ignore_topics">[]</rosparam>
    <rosparam param="ignore_publishers">[]</rosparam>
    <rosparam param="ignore_subscribers">[]</rosparam>

    <!-- The list of topics to synchronize. The ~ignore_nodes and ~ignore_topics lists are processed first. -->
    <rosparam param="sync_topics">['/pickup_command',  '/pickup_response']</rosparam>

    <!-- The list of services, which are to be ignored while a synchroniation. The ~ignore_nodes list is processed first.
    These are defaults and cannot be removed: ['/*get_loggers', '/*set_logger_level'] -->
    <rosparam param="ignore_services">[]</rosparam>

    <!-- The list of services to synchronize. The ~ignore_nodes and ~ignore_services lists are processed first. -->
    <rosparam param="sync_services">[]</rosparam>

    <!-- This parameter is only regarded, if sync_nodes or sync_topics are set. In this case the subscribed and published topics are synchronized even if they are not in the sync_* list -->
    <param name="sync_topics_on_demand" value="False" />

    <!-- The list of types to ignore.
    These are defaults and cannot be removed: ['bond/Status'] -->
    <rosparam param="ignore_type">[]</rosparam>

    <!-- The nodes which are running not at the same host as the ROS master are not synchronized by default. Use sync_remote_nodes to sync these nodes also. -->
    <param name="sync_remote_nodes" value="False" />
  </node>
</launch>
```

atiderko, 2017-05-29:
Hi @BasJ93,

what do you mean with 'crashes'? Will it be shutdown by ROS or is there system failure?
are the names of the ROS Nodes provides the topics `'/pickup_command',  '/pickup_response'` equal?

can you add 
`<param name="log_level" value="DEBUG" />`
to master_sync and lock at the output. There are listed all synchronisations.

BasJ93, 2017-05-31:
Hi @atiderko 

I did not have time to test yesterday. However I just tested again, and the error I get about the capability_server is:
REQUIRED process [capability_server-9] has died!

After this happens, of course roscore shuts down.

However, I do think the nodes have the same name. I thought they would get a random appendix when they launch, but rosnode list shows that they use the same names on both robots. I presume this is a problem?

atiderko, 2017-05-31:
the same node name is definitely a problem. Can you try to add the capability service to ignore nodes?

BasJ93, 2017-07-11:
I found my problem can be solved by namespacing the robots. This issue can be closed.

closed 2017-07-11 by BasJ93