fkie/multimaster_fkie #64: Ignoring TF sync

akifh, 2017-09-25:
Hello,

Think that we have two robots with their topics and nodes are namespaced. I want to ignore `/tf` topic since tf information on another robot is not relevant to the other. I have tried to add `/tf` to `ignore_topics` parameter however this does not seem to work. Still TF information is received from the other robot. Can you guide me?

Thanks

atiderko, 2017-09-25:
hi,

did you started it with `rosrun master_sync_fkie master_sync _ignore_topics:=[/tf]`?
The output should contain something like:
```
ignore_topics: ['/rosout', '/tf', '/rosout_agg']
```
If not, can you provide the console output of master_sync node, please.

akifh, 2017-09-25:
@atiderko, seems like there was a problem with master_sync namespace. It is working now. Thanks.

closed 2017-09-25 by akifh