fkie/multimaster_fkie #66: ROS Parameter Server Sync

baronep, 2017-11-13:
I saw on the github page that the parameter server is not synced with the current version of the software. What was the reason for this? Is there a blocking issue on this?

atiderko, 2017-11-15:
it was a decision to not implement this feature. Key arguments are:
* a lot of parameter are used by nodes only on start. 
* the dynamic reconfigure changes the parameter on the ROS parameter server where the ROS node is registered. You need for these parameter also no synchronization.
* not clear collision resolution of global parameter
* parameter with huge amount of data, e.g. robot_description, can lead to unnecessary network load, so it is better to define this locally again.

If you still need parameter synchronization, there is one in a [fork](https://github.com/jhu-lcsr-forks/multimaster_fkie/tree/cc35c4ed8ce1f209274fe01554f0067fac3cc715/param_sync_fkie) with parameter sync node

atiderko, 2018-12-12:
added separate script [param_sync.py](https://github.com/fkie/multimaster_fkie/blob/master/master_sync_fkie/nodes/param_sync.py), see  commit [b5805b3](https://github.com/fkie/multimaster_fkie/commit/b5805b3963730d352baec7436fb04f694f137aa8)

closed 2018-12-12 by atiderko

marcus011, 2021-07-28:
Hi @atiderko ,
How to use this param_sync file ,can you explain more about it

atiderko, 2021-07-30:
Hi @marcus011,
an example of the .sync file can be found in the launch folder of the `fkie_master_sync`. Es ist eine ROS-format compatible YAML-Datei mit den gleichen Parametern wie in der Launch-Datei. It can be loaded in the launch file using either the `rosparam` or the` ~interface_url` parameter.

The `~interface_url` parameter was introduced for use with Node Manager. On launch the `master_sync` using the button in [host description panel](http://fkie.github.io/multimaster_fkie/chapter_usage/host_description_panel.html) you can choose a .sync file.

marcus011, 2021-07-30:
Hi @atiderko ,
Thanks for the information, basically I want to load /robot_description parameter from another roscore using a launch file to my gazebo roscore,
so can you guide how I can sync those parameter to my gazebo ,so that robot can be loaded up

atiderko, 2021-07-30:
On the machine where you need remote parameter run:
`rosrun fkie_master_sync param_sync.py`

It is a simple script which listen to changes published by `master_discovery` and get parameter from remote `roscore`.

marcus011, 2021-07-30:
I have tested that command & robot_description is not loading up on gazebo side

atiderko, 2021-07-30:
Sorry, I see now, a namespace is added by default. Try this:
`rosrun fkie_master_sync param_sync.py _add_ns:=False`

marcus011, 2021-07-30:
I am getting this error when using command
`rosrun fkie_master_sync param_sync.py _add_ns:=False`
```
[ERROR] [1627628967.414029]: bad callback: <function master_changed at 0x7f0eb5c1e2d0>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 748, in _invoke_callback
    cb(msg, cb_args)
  File "/home/amantya002/catkin_ws/src/multimaster/fkie_master_sync/nodes/param_sync.py", line 32, in master_changed
    master_to['/'+_ns] = params_from
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msproxy.py", line 138, in __setitem__
    rospy.impl.paramserver.get_param_server_cache().update(resolved_key, val)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/paramserver.py", line 102, in update
    value_key = namespaces[-1]
IndexError: list index out of range

```

atiderko, 2021-07-30:
hmm... currently I have no time to debug.
Perhaps something is wrong with provided namespaces to MasterProxy. The script is very easy if you want to try to fix ;-)