fkie/multimaster_fkie #71: unable to detect the other master using master_discovery in one computer while the other computer can detect my master

Mbadra96, 2018-02-22:
##There is no errors 
# the first copmuter with ip 192.168.1.6
badra@badra-VirtualBox:~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
[INFO] [1519305154.375028]: ROS Master URI: http://192.168.1.6:11311/
[INFO] [1519305154.413346]: Check the ROS Master[Hz]: 1
[INFO] [1519305154.413815]: Heart beat [Hz]: 0.02
[INFO] [1519305154.418971]: Active request after [sec]: 60
[INFO] [1519305154.419649]: Remove after [sec]: 300
[INFO] [1519305154.420142]: Robot hosts: []
[INFO] [1519305154.420636]: Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [1519305154.552176]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1519305154.554323]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1519305154.571384]: Create multicast socket at ('224.0.0.1', 11511)
[INFO] [1519305154.790963]: Detected master discovery: http://localhost:11611
[INFO] [1519305154.898284]: Added master with ROS_MASTER_URI=http://badra-VirtualBox:11311/

# The second computer with ip 192.168.1.8
nasr@nasr-VirtualBox:~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
[INFO] [1519305169.413408]: ROS Master URI: http://192.168.1.8:11311/
[INFO] [1519305169.437923]: Check the ROS Master[Hz]: 1
[INFO] [1519305169.438593]: Heart beat [Hz]: 0.02
[INFO] [1519305169.444326]: Active request after [sec]: 60
[INFO] [1519305169.445029]: Remove after [sec]: 300
[INFO] [1519305169.445306]: Robot hosts: []
[INFO] [1519305169.447084]: Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [1519305169.495208]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1519305169.501487]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1519305169.530136]: Create multicast socket at ('224.0.0.1', 11511)
[INFO] [1519305169.754659]: Detected master discovery: http://localhost:11611
[INFO] [1519305169.858051]: Added master with ROS_MASTER_URI=http://nasr-VirtualBox:11311/
[INFO] [1519305205.267947]: Detected master discovery: http://192.168.1.6:11611
[INFO] [1519305205.872249]: Added master with ROS_MASTER_URI=http://badra-VirtualBox:11311/

i have add the ip on the etc/hosts and the export ROS_Master_URI on the bashrc 
and the two computers can ping each other with no errors

atiderko, 2018-02-22:
please try this:
```
badra@badra-VirtualBox:~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1 _heartbeat_hz:=0.5
nasr@nasr-VirtualBox:~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1 _heartbeat_hz:=0.5
```
If this does not help try following:
```
badra@badra-VirtualBox:~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1 _heartbeat_hz:=0.5 _robot_hosts:=[nasr-VirtualBox]
nasr@nasr-VirtualBox:~$ rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1 _heartbeat_hz:=0.5 _robot_hosts:=[badra-VirtualBox]
```

Mbadra96, 2018-02-22:
first Thank you for your quick response .
the second method works.
may i know if this the only way to make it work?
what was the problem ?

atiderko, 2018-02-22:
Second method uses unicast communication to defined robot hosts.
I guess the multicast routing from `nasr-VirtualBox` to `badra-VirtualBox` does not work properly.
You can try to check the routing tables `route -n`.

Mbadra96, 2018-02-22:
Thank you i will check it

closed 2018-03-18 by Mbadra96