fkie/multimaster_fkie #72: Sharing pose data via multimaster between three robots

sjoshi30, 2018-04-17:
I want to run AMCL package on three different robots(each robot has laptop with ros running on it) and share the pose data via multimaster_fkie. When we sync the three laptops, the bandwidth decreases significantly. Is there any solution for this ?

atiderko, 2018-04-17:
Hi,

you can add filter to exclude some topics from synchronization. E.g. /tf topic is a candidate. Add follow parameter to *master_sync* node:
``` <rosparam param="ignore_topics">[/tf]</rosparam>```

For more filter options see [master_sync.launch](https://github.com/fkie/multimaster_fkie/blob/master/master_sync_fkie/launch/master_sync.launch)

sjoshi30, 2018-04-22:
Great, thank you for your response! It worked :)

closed 2018-04-27 by atiderko