fkie/multimaster_fkie #74: Connection refused in starting master_discovery_fkie

coolkesword, 2018-05-26:
Dear sir/madam,

Hi, I am currently using the multimaster_fkie to setup the intercommunication between several robots, and some problems occurred in launching the multimaster_fkie package. 

The software environment is: Ubuntu 16.04 + ROS kinetic + multimaster_fkie kinetic-devel branch. The hardware setup is described as follows: we use a TPlink router to build up a wireless network, and each robot is equipped with a wireless interface card (DLink) and connected to the router. 

We followed exactly the procedure as described in the section 3 of the document: http://www.iri.upc.edu/files/scidoc/1607-Multi-master-ROS-systems.pdf. Each robot can ping 244.0.0.1 successfully, but the package cannot be launched and the error message is shown below. Could you kindly advise on how to proceed? Is it related with our hardware setting or software configuration? We also tried parameter settings like interface, send_mcast and robot_hosts but they didn't help. Thanks a lot in advance!

rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
Error while set the log level: [Errno 111] Connection refused
->INFO level will be used!
Traceback (most recent call last):
  File "/home/fh/catkin_ws/src/multimaster_fkie/master_discovery_fkie/nodes/master_discovery", line 4, in <module>
    master_discovery_fkie.main()
  File "<string>", line 97, in main
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 326, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 514, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1053, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 897, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 859, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 836, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
socket.error: [Errno 111] Connection refused

Thanks and regards
Jie

atiderko, 2018-05-26:
Hello Jie,

I get the same error if __roscore__ is not running. If you try to start _master_discovery_ without passed arguments you will see the warning about not running ROS master.
It seems to be an unhandled error in ROS.

coolkesword, 2018-05-26:
Dear Mr. Atiderko, 

Thanks for your reply. We had run roscore before launching multimaster fkie package, so roscore is not the cause for our problem. It's worthy to mention that things turned out to work successfully if we equip our robots with computers with integrated wireless interface cards. Hence the problem should be related with the standalone wireless interface cards. 

Regards
Jie