fkie/multimaster_fkie #77: demo_capabilities.launch

JadBatmobile, 2018-07-30:
When such a launch file is executed from demo_bob.launch, the following WARNING message:

WARNING: [/home/hallab/multi-master/src/multimaster_fkie/node_manager_fkie/launch/demo_capabilities.launch] unknown <rosparam> attribute 'tag'

And nothing appears in the capability view. I have yet to get that view to work for me.

atiderko, 2018-07-31:
Please load the `demo_bob.launch` using `default` button in the [launch dog](http://fkie.github.io/multimaster_fkie/chapter_usage/launch_dock.html). Only launch files loaded as default configuration will be displayed in [capability view](http://fkie.github.io/multimaster_fkie/chapter_usage/capability_view.html)

atiderko, 2018-12-10:
I hope, the issue is solved.
If not, please reopen!

closed 2018-12-10 by atiderko