fkie/multimaster_fkie #79: node_manager crashes on startup due to missing MessageFrame.ui

samlcharreyron, 2018-08-10:
Since updating node_manager_fkie from 0.7.8 to 0.8, I can no longer open the node manager without it crashing. The traceback is as shown below.

```
[INFO] [1533888798.773351]: listen for logs on /rosout
[INFO] [1533888799.061694]: Start RPC-XML Server at ('0.0.0.0', 22622)
[INFO] [1533888799.062469]: Subscribe to parameter `/roslaunch/uris`
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/node_manager_fkie/main_window.py", line 779, in on_master_state_changed
    self.getMaster(msg.master.uri).master_state = msg.master
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/node_manager_fkie/main_window.py", line 572, in getMaster
    self.masters[masteruri] = MasterViewProxy(masteruri, self)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/node_manager_fkie/master_view_proxy.py", line 358, in __init__
    self.message_frame = MessageFrame()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/node_manager_fkie/message_frame.py", line 137, in __init__
    loadUi(ui_file, self.frameui)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 247, in loadUi
    return _loadUi(uifile, baseinstance, custom_widgets)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 151, in _loadUi
    return uic.loadUi(uifile, baseinstance=baseinstance)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/__init__.py", line 222, in loadUi
    return DynamicUILoader(package).loadUi(uifile, baseinstance, resource_suffix)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/Loader/loader.py", line 72, in loadUi
    return self.parse(filename, resource_suffix, basedir)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 972, in parse
    document = parse(filename)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 1182, in parse
    tree.parse(source, parser)
  File "/usr/lib/python2.7/xml/etree/ElementTree.py", line 647, in parse
    source = open(source, "rb")
IOError: [Errno 2] No such file or directory: '/opt/ros/kinetic/lib/python2.7/dist-packages/node_manager_fkie/MessageFrame.ui'
```

atiderko, 2018-08-10:
Thank you for reporting and sorry for trouble!
I released a new version, which should fix this problem.

atiderko, 2018-12-10:
I hope, the issue is solved.
If not, please reopen!

closed 2018-12-10 by atiderko