fkie/multimaster_fkie #80: Help: sync /tf topic to another machine cause BufferCore::getLatestCommonTime get wrong time?

zhouchengming1, 2018-09-09:
Hello, all:

I use multimaster to sync some topics and nodes for two machines, but I found if I sync /tf topic from A to B machine , then the program on A will get wrong time from BufferCore::getLatestCommonTime.
I don't know how it happens, anyone have idea?
Thanks.

atiderko, 2018-09-09:
Hi,

did you synchronize time between machines A and B?
http://wiki.ros.org/ROS/NetworkSetup#Timing_issues.2C_TF_complaining_about_extrapolation_into_the_future.3F

zhouchengming1, 2018-09-09:
@atiderko  No, I will try this. But I don't know why /tf topic synced to B machine could cause a problem like this? Does it get some time information from B when register?

zhouchengming1, 2018-09-10:
Does the multimaster sync node will modify the timestamp of local ros master ?

atiderko, 2018-09-10:
`master_sync` does not modify any timestamps or other data or behaviour of ROS communication.
If you synchronize /tf (and also other topics) the sender stay the same and publish tf messages from different hosts. The receiver of /tf get messages with different timestamps.
I do not know if it will cause the BufferCore :: getLatestCommonTime error. It is just a guess!

zhouchengming1, 2018-09-14:
It seems that two rosnodes A and B are synced to another machine, A publish to /tf topic and B subscribe to /tf topic , so another data connection created on the remote machine?
When  I sync only needed topics, instead of the whole nodes, the problem is gone.

closed 2018-09-14 by zhouchengming1