fkie/multimaster_fkie #83: Organize nodes view by namespaces

acschaefer, 2018-10-25:
Thank you for this great piece of software! 

I have been using `node_manager` for some time now. In large projects, nodes are often organized in namespaces, for example
```
ground_robot/lidar/velodyne_driver
ground_robot/vision/camera_front_driver/rgb_rectifier
ground_robot/vision/camera_front_driver/nir_rectifier
ground_robot/camera_back_driver
aerial_robot/flight_controller
aerial_robot/vision/camera_1
aerial_robot/vision/camera_2
...
```
That is what the `node_manager`'s node view looks like. However, I find it hard to get an overview over all the nodes, if I have a couple of dozen running. So why not use the namespace topology to create a nice nested node view like this:
```
ground_robot/
    lidar/
        velodyne_driver
    vision/
        camera_front_driver/
            rgb_rectifier
            nir_rectifier
        camera_back_driver
aerial_robot/
    flight_controller
    vision/
        camera_1
        camera_2
...
```

Capability grouping is not really an alternative. I would end up with two levels only, and I would need to iterate over all nodes and add the `capability_group` parameter to each of them. If the topology of the namespace changed, I would have to change all `capability_group` tags again.

atiderko, 2018-10-29:
Nice to hear that the `node manager` is also useful to others!

We didn't need the grouping by namespaces, because we use only two level :-)
But it is good point for all others with deeper namespacing. I will put  this suggestion on my todo!

atiderko, 2018-10-31:
I added a branch `test_nsgroup` with new grouping. Can you take a look on the `node manager`. Is it what you imagine?

acschaefer, 2018-11-03:
Thank you for reacting so quickly! I checked out the test_nsgroup branch and reinstalled the node manager. However, I cannot see any changes in the nodes view compared to the previous version (see screenshot below).

Hopefully, you can reproduce this error by loading the little test launch file I wrote. It is contained in my [pull request](https://github.com/fkie/multimaster_fkie/pull/84).

[image]

atiderko, 2018-11-03:
Thank you for testing!

But I can not reproduce your issue, see image:
[image]

I removed a warning while loading this file and updated a version tag. After update you should see the same node manager version if you click on info button. Only to see if the same version is started ;-)

acschaefer, 2018-11-06:
Yes, you were right, I was still using the old version. With your new version, it worked like a charm! =) I also added some parameters to the dummy nodes in the test script in order to make sure they start up alright. They do: 

[image]

acschaefer, 2018-11-06:
This new feature makes it a lot easier to get an overview over large ROS systems.

I think namespace grouping could be a good idea for topics and services as well. Would the changes you made to the nodes view be easily transferable to the topics and services views?

As a motivational example, here is the beginning of my current topics view:

[image]

atiderko, 2018-11-07:
good to hear that it works.

Unfortunately it is not so easily to transfer, but it is also on my todo!
I will merge this changes into the master branch after topic and service view is changed.

In the meantime you can test the topic view and new group/node highlighting ;-)

acschaefer, 2018-11-08:
Sounds great! I am testing every day. =)

One minor remark: Should namespaces that contain only one node be opened automatically in the nodes view?

atiderko, 2018-11-09:
I made some changes:
- no groups with one node are created
- namespace groups for topics
- namespace groups for services

acschaefer, 2018-11-14:
I have been testing for the last few days and it worked great! 

Just one minor remark: As all topics and services start with a slash, I always see a single `/` as top node. I suggest to ignore the slash prefix when creating namespace groups.

I attached a screenshot for illustration: 
[image]

atiderko, 2018-11-14:
thank you for testing!

Since I can not easily replace the root item of the TreeView I changed the name of `/`

Can you try again, please?

acschaefer, 2018-11-15:
Thank you for updating the code so quickly!

The screenshot below shows the services view after updating to the most recent version of node_manager. The slash at the top is still there. Moreover, I do not understand the subdivision of services into `/` and `{SYSTEM}` yet. Many services are listed as system services which I would not consider being core ROS services.

I hope this is helpful. =)

[image]

atiderko, 2018-11-15:
I put `get_loggers` and `set_logger_level` to `{SYSTEM}` group since this services are created by core ROS for each node and you did not actively use them.

The same is for topics: `rosout` and `rosout_agg` are always there, so I put them to `{SYSTEM}` group.

`/` in services  was also changed, but in the second commit. Perhaps you use not the newest commit?

acschaefer, 2018-11-16:
Sorry, it happened again! =) With catkin, you can have an installed version that is different from the built, but not installed version.

So I upgraded to your newest version and it works perfectly! I also understood the distinction between system services and other services. That makes sense.

Thank you for this nice extension!

acschaefer, 2018-11-16:
Here comes some more feedback from testing the namespace grouping.

I have a project with about 20 nodes and around 100 topics. When I select a node in the node view and then try to change to the services view, the node_manager crashes reproducibly every time.

atiderko, 2018-11-16:
Thanky for testing!

I fixed the bug on change to service view.

acschaefer, 2018-11-17:
Confirmed! The bug is gone. I really like the namespace grouping.

acschaefer, 2018-11-20:
I found a bug in the topics view: The publisher/subscriber counts are not correct. Take a look at the following screenshot. Although `robot/joint_states` is published by Gazebo as shown in the panel on the left-hand side, the publisher count for the topic displayed in the main panel is 0. 
[image]

atiderko, 2018-11-20:
Thank you for reporting! I fixed this issue.

acschaefer, 2018-11-28:
Confirmed, it is fixed!

atiderko, 2018-11-28:
thank you for confirmation!

I merged this branch into the master branch.

closed 2018-11-28 by atiderko