fkie/multimaster_fkie #89: Image transport

mrsp, 2018-12-06:
Hello

When I subscribe a synced image topic of type sensor_msgs/Image the rate drops dramatically with every subscription!

More specifically, I sync two ROS masters via ethernet (100Mb/s), the first ROS Master sends an image over ethernet to another ROS master where I subscribe to get the Image. The rate  drops from 40hz (which is the rate before transmition via multimaster_fkie) to ~10 hz. 

Any ideas ? Is this a bug? Since this rate drop happens only in the image topics!

Thanks in advance!

atiderko, 2018-12-06:
Hi

multimaster_fkie does not transport any topic data. It only register the topic data on ROS Master and ROS does all other magic stuff. This means that the Image data is transmitted using the ROS transport protocol (TCP or UDP).

You can try to subscribe CompressedImage on remote host to reduce the load on network. See [teminal treffic load](https://askubuntu.com/questions/257263/how-to-display-network-traffic-in-the-terminal)

mrsp, 2018-12-06:
I see!

Thanks for the quick reply

closed 2018-12-10 by atiderko