fkie/multimaster_fkie #90: master_sync gets stuck in timestamp loop

Cartoonman, 2018-12-07:
I have two devices running multimaster_fkie with sync and discovery. local host is slave to the remote host's ntp server, and both are time synced. I'm using the zeroconf discovery method. 

I seem to have stumbled across a strange race condition where the remote host gets to a state where it doesn't require a request to the local host, but the local host gets stuck in a request loop to the remote host.

Specifically lines 433-437 in sync_thread.py is where the crux of the problem occurs. The timestamps timestamp_remote and stamp_local do not change with each request, so it just keeps sending RPC requests over the network. 

My guess is that there seems to be a problem in the way master state and timestamps are updated with zeroconf, since I have seen that if I specifically monitor zeroconf/changes through the node_manager, the remote host's timestamp in sync_thread does get updated to the latest one after a request (and stays constant thereafter), but the local host's timestamp remains the same (as reported by the remote host). 

I've been able to somewhat reproduce the issue by staging a launch file through default config launch on the local host and monitoring master_sync's output, as well as over Wireshark.

atiderko, 2018-12-07:
I did not tested zeroconf method for long time...
Do you have the debug output of `master_sync`, especially with lines: 433 and 435 in  sync_thread.py

You can add `/master_sync/log_level` parameter with `DEBUG` value and restart master_sync to see all debug messages since start:
`$ rosparam set /master_sync/log_level DEBUG`

I will try to reproduce the problem!

Cartoonman, 2018-12-07:
Yes, here's a snippet. I have two remote devices but one of them does not have the sync node on.

[32m[DEBUG] [1544204739.649943]: init_node, name[/master_sync], pid[29102]
[0m[32m[DEBUG] [1544204739.651098]: binding to 0.0.0.0 0
[0m[32m[DEBUG] [1544204739.651697]: bound to 0.0.0.0 34495
[0m[32m[DEBUG] [1544204739.652648]: ... service URL is rosrpc://localmachine:34495
[0m[32m[DEBUG] [1544204739.653185]: [/master_sync/get_loggers]: new Service instance
[0m[32m[DEBUG] [1544204739.656799]: ... service URL is rosrpc://localmachine:34495
[0m[32m[DEBUG] [1544204739.657328]: [/master_sync/set_logger_level]: new Service instance
[0m]2;master_sync[INFO] [1544204739.672933]: ignore_hosts: []
[INFO] [1544204739.676031]: sync_hosts: []
[INFO] [1544204739.681605]: sync_topics_on_demand: False
[INFO] [1544204739.684349]: resync_on_reconnect: True
[INFO] [1544204739.687261]: resync_on_reconnect_timeout: 0
[INFO] [1544204739.699292]: listen for updates on /zeroconf/changes
[32m[DEBUG] [1544204739.705786]: ... service URL is rosrpc://localmachine:34495
[0m[32m[DEBUG] [1544204739.706325]: [/master_sync/get_sync_info]: new Service instance
[0m[32m[DEBUG] [1544204739.725828]: connecting to localmachine 43593
[0m[32m[DEBUG] [1544204739.728413]: connecting to localmachine 43593
[0m[INFO] [1544204739.731472]: ROS masters obtained from '/zeroconf/list_masters': ['localmachine', 'remote-host-2', 'remote-host-1']
[32m[DEBUG] [1544204739.732141]: SyncThread[remote-host-2]: create this sync thread
[0m[INFO] [1544204739.739170]: [remote-host-2] ignore_nodes: ['/master_sync', '/node_manager', '/rosout', '/zeroconf']
[INFO] [1544204739.744759]: [remote-host-2] sync_nodes: []
[INFO] [1544204739.749773]: [remote-host-2] ignore_topics: ['/rosout', '/rosout_agg']
[INFO] [1544204739.755276]: [remote-host-2] sync_topics: []
[INFO] [1544204739.760367]: [remote-host-2] ignore_services: ['/*get_loggers', '/*set_logger_level']
[INFO] [1544204739.765872]: [remote-host-2] sync_services: []
[INFO] [1544204739.771228]: [remote-host-2] ignore_type: ['bond/Status']
[INFO] [1544204739.776739]: [remote-host-2] ignore_publishers: []
[INFO] [1544204739.781756]: [remote-host-2] ignore_subscribers: []
[32m[DEBUG] [1544204739.790722]: SyncThread[remote-host-2]: update notify new timestamp(1544194679.901128054), old(0.000000000)
[0m[32m[DEBUG] [1544204739.791761]: SyncThread[remote-host-1]: create this sync thread
[0m[INFO] [1544204739.800109]: [remote-host-1] ignore_nodes: ['/master_sync', '/node_manager', '/rosout', '/zeroconf']
[INFO] [1544204739.805337]: [remote-host-1] sync_nodes: []
[INFO] [1544204739.810586]: [remote-host-1] ignore_topics: ['/rosout', '/rosout_agg']
[INFO] [1544204739.814409]: [remote-host-1] sync_topics: []
[INFO] [1544204739.817172]: [remote-host-1] ignore_services: ['/*get_loggers', '/*set_logger_level']
[INFO] [1544204739.820132]: [remote-host-1] sync_services: []
[INFO] [1544204739.822310]: [remote-host-1] ignore_type: ['bond/Status']
[INFO] [1544204739.824511]: [remote-host-1] ignore_publishers: []
[INFO] [1544204739.826416]: [remote-host-1] ignore_subscribers: []
[32m[DEBUG] [1544204739.829017]: SyncThread[remote-host-1]: update notify new timestamp(1544198997.926506042), old(0.000000000)
[0m[32m[DEBUG] [1544204739.829589]: node[/master_sync, http://localmachine:38591/] entering spin(), pid[29102]
[0m[INFO] [1544204740.294743]: SyncThread[remote-host-1] Requesting remote state from 'http://remote-host-1:11911/'
[INFO] [1544204740.301757]: SyncThread[remote-host-1] Applying remote state...
[32m[DEBUG] [1544204740.305221]: SyncThread[remote-host-1]: current timestamp 1544198997.926506042, local 1544196658.900779963
[0m[32m[DEBUG] [1544204740.306395]: SyncThread[remote-host-1]: invoke next update, remote ts: 1544198997.926506042
[0m[INFO] [1544204740.308146]: SyncThread[remote-host-1] remote state applied.
[INFO] [1544204740.776493]: SyncThread[remote-host-1] Requesting remote state from 'http://remote-host-1:11911/'
[INFO] [1544204740.783588]: SyncThread[remote-host-1] Applying remote state...
[32m[DEBUG] [1544204740.787265]: SyncThread[remote-host-1]: current timestamp 1544198997.926506042, local 1544196658.900779963
[0m[32m[DEBUG] [1544204740.788438]: SyncThread[remote-host-1]: invoke next update, remote ts: 1544198997.926506042
[0m[INFO] [1544204740.789904]: SyncThread[remote-host-1] remote state applied.
[INFO] [1544204741.439328]: SyncThread[remote-host-2] Requesting remote state from 'http://remote-host-2:11911/'
[INFO] [1544204741.444130]: SyncThread[remote-host-2] Applying remote state...
[32m[DEBUG] [1544204741.445528]: SyncThread[remote-host-2]: current timestamp 1544194679.901128054, local 1544194679.901128054
[0m[INFO] [1544204741.445939]: SyncThread[remote-host-2] remote state applied.
[INFO] [1544204741.809589]: SyncThread[remote-host-1] Requesting remote state from 'http://remote-host-1:11911/'
[INFO] [1544204741.816672]: SyncThread[remote-host-1] Applying remote state...
[32m[DEBUG] [1544204741.820567]: SyncThread[remote-host-1]: current timestamp 1544198997.926506042, local 1544196658.900779963
[0m[32m[DEBUG] [1544204741.821615]: SyncThread[remote-host-1]: invoke next update, remote ts: 1544198997.926506042
[0m[INFO] [1544204741.823084]: SyncThread[remote-host-1] remote state applied.
[INFO] [1544204742.950957]: SyncThread[remote-host-1] Requesting remote state from 'http://remote-host-1:11911/'
[... cut ...]
[INFO] [1544204767.465325]: SyncThread[remote-host-1] Requesting remote state from 'http://remote-host-1:11911/'
[INFO] [1544204767.472247]: SyncThread[remote-host-1] Applying remote state...
[32m[DEBUG] [1544204767.475803]: SyncThread[remote-host-1]: current timestamp 1544198997.926506042, local 1544196658.900779963
[0m[32m[DEBUG] [1544204767.476908]: SyncThread[remote-host-1]: invoke next update, remote ts: 1544198997.926506042
[0m[INFO] [1544204767.478322]: SyncThread[remote-host-1] remote state applied.

atiderko, 2018-12-07:
I added a fix to avoid request loop. I hope it this solve the problem.

Long time ago I added a new timestamp for changes which do not regard the changes caused by master_sync to avoid recyncs. I added it only to master_discovery :-/

Thank you for reporting this issue!

Cartoonman, 2018-12-07:
Hotfix works! Thanks for your very fast reply and fix!

Cheers!

closed 2018-12-07 by Cartoonman