fkie/multimaster_fkie #93: Running 2 Masters with ROS_MASTER_URI of localhost

Gil401, 2019-01-27:
Hi Guys,

So I am trying to have 2 masters on 2 different machines (one master on every device) so that once I have a connectivity failure / loss on a device, the master will keep running with it's local nodes (without communicating with the other master).
I was thinking to achieve this by setting the ROS_MASTER_URI on both devices to localhost, however, it seems that it does not work.

Is there a way to achieve it?
Thanks,

atiderko, 2019-01-28:
Hi Gil401,

you have to set the ROS_MASTER_URI to the public hostname of your machine.
`localhost` in ROS_MASTER_URI is only for one host systems, see
http://wiki.ros.org/ROS/EnvironmentVariables#ROS_MASTER_URI

regards

Gil401, 2019-01-28:
Hi @atiderko, thank you for the response.

Is there a way to workaround it or a different way to achieve what I am looking for?

atiderko, 2019-03-14:
I hope you found a solution. If you have further questions about this topic, you can reopen this issue!

closed 2019-03-14 by atiderko

nikv96, 2019-07-01:
Hi @atiderko, do you have any suggestions for external WiFi receivers that you've tested? I am facing the same trouble and can't seem to find a work around.

atiderko, 2019-07-03:
Hi @nikv96, we use Ubiquiti Bullet for WiFi as one cheap solution. Other solutions we tested are more expensive.