fkie/multimaster_fkie #94: Masters are discovered but without the nodes

Gil401, 2019-02-05:
Hi,
I have 2 devices.
After starting roscore on them both and a few of my nodes, I have executed the following:

Station Device:
```
rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
rosrun master_sync_fkie master_sync __name:=station
```

Robot Device:

```
rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
rosrun master_sync_fkie master_sync __name:=robot
```

`rosservice call /master_discovery/list_masters `
> masters: 
>   - 
>     name: "robot"
>     uri: "http://robot:11311/"
>     timestamp: 1549369438.64
>     timestamp_local: 1549369115.78
>     online: False
>     discoverer_name: "/master_discovery"
>     monitoruri: "http://10.10.80.6:11611"
>   
>     name: "station"
>     uri: "http://station:11311/"
>     timestamp: 1549369766.92
>     timestamp_local: 1549369766.92
>     online: True
>     discoverer_name: "/master_discovery"
>     monitoruri: "http://localhost:11611"

however when running `rosnode list` / `rostopic list` / `rosservice list` I do not see any data from the other device.

What am I missing?

atiderko, 2019-02-05:
Hi,

you renamed `master_sync` node, thus it synchronized the remote `master_sync` too.
Please remove `__name:=station` and `__name:=robot` or add it to ignore list:

`rosrun master_sync_fkie master_sync __name:=station _ignore_nodes:=[/robot]`
`rosrun master_sync_fkie master_sync __name:=robot _ignore_nodes:=[/station]`

Gil401, 2019-02-06:
Hi Alexander,

Thank you for the quick response.
Now it seems to work, however I do not see all the nodes when I am running rosnode list.

For example I have a launch file where I have a manager with 2 nodes in it:

```
<launch>
	<arg name="manager" default="indoor_device_manager_standalone_nodelet_station"/>
	<node pkg="nodelet" type="nodelet" name="$(arg manager)" args="manager" output="screen"/>
	<node pkg="nodelet" type="nodelet" name="indoor_device_manager_station" args="load indoor_device_manager/IndoorDeviceManager $(arg manager)" required="true">
	</node>
	<node pkg="indoor_device_manager" name="station_topic_health_monitor" type="topic_hz.py" required="true">
	</node>

</launch> 
 
```
but when I do `rosnode list` I see only
`/indoor_device_manager_standalone_nodelet_station` (which is the name of the manager) Is there something I can do?

atiderko, 2019-02-06:
Hi Gil401,

only topics and services can be registered on ROS-Master. If a ROS-Node does not contain any topics or services it is also not in the `rosnode list`.
All topics of the nodelets are assigned to the nodelet-manager. On the origin ROS-Master the nodelets are listed because they subscribe the topic of type `bond/Status`.* This topic will not be synchronized to avoid broken nodelets. This is why you no see nodelets-nodes on remote (synchronized) master. But your system should work anyway.
Is there any reason why you need to see remote nodes on local system? If you need to manage nodes in your multi-host system you can use `node_manager` of multimaster package.

*There are some other topics, which are also not synchronized. You can see them all in the output of `master_sync`.

Gil401, 2019-02-06:
> _Is there any reason why you need to see remote nodes on local system?_

Yes, because I am monitoring the system and checking for certain nodes. Is there a way to get the nodes from the node_manager via an api / terminal?

Thank you very much!

atiderko, 2019-02-06:
You can use the XML-RCP interface of `master_dicovery`:
http://wiki.ros.org/master_discovery_fkie?distro=melodic

`masterInfo()` reports current state of ROS-master where `master_discovery` is running.
By default it listens on port `11611`.

closed 2019-03-14 by atiderko