fkie/multimaster_fkie #95: MasterMonitor[/move_base/global_costmap/static_layer/set_parameters]: can't get contact information. ROS master responds with: no provider

sunil1349, 2019-02-20:
MasterMonitor[/move_base/global_costmap/static_layer/set_parameters]: can't get contact information. ROS master responds with: no provider

i have two robots running on same floor i am trying to plan path form another computer suggest how to do it

atiderko, 2019-02-20:
the error says that the service and therefor the move_base node not longer available, not running or not reachable.
Did you run `/move_base` on multiple hosts? And synchronized (by `master_sync`)? If so, one of the node with same name was stopped by ROS.

For suggestions you have to provide a bit more  informations ;-)

sunil1349, 2019-02-20:
i have same type ros nodes running in two robots (not two maybe more), in both robots i have same path planning topics , i want run path planning in another system so that to robots can know each other where there are moving ,also can avoid clash of paths

sunil1349, 2019-02-20:
in one of research papers they said multimaster_fkie is useful for it

sunil1349, 2019-02-20:
`<launch>
  <arg name="robot_name" default="robot0" />
  <node name="master_discovery" pkg="master_discovery_fkie" type="master_discovery" >
  </node>
  <node name="master_sync" pkg="master_sync_fkie" type="master_sync">
  </node>
  
  <node pkg="topic_tools" type="relay" name="robot_relay" args="/shared_topic /$(arg robot_name)/shared_topic" />
</launch>`
in above code master_sync node how can i mention shared_topics or nodes

sunil1349, 2019-02-20:
<launch>
  <arg name="robot_name" default="robot0" />
  <node name="master_discovery" pkg="master_discovery_fkie" type="master_discovery" >
  </node>
  <node name="master_sync" pkg="master_sync_fkie" type="master_sync" >
  </node>
  
  <node pkg="topic_tools" type="relay" name="robot_relay" args="/shared_topic /$(arg robot_name)/shared_topic" />
</launch>

how can i give my shared topics or nodes to sync

sunil1349, 2019-02-21:
/all_ranges
/amcl/parameter_descriptions
/amcl/parameter_updates
/amcl_pose
/battery_state
/bumper_hit
/cmd_vel
/cmd_vel_joy
/cmd_vel_move_base
/diagnostics
/front_bottom_left_sonar
/front_bottom_middle_sonar
/front_bottom_right_sonar
/front_scan
/front_top_left_sonar
/front_top_right_sonar
/laserscan_multi_merger/parameter_descriptions
/laserscan_multi_merger/parameter_updates
/map
/parameter_descriptions
/move_base/TebLocalPlannerROS/global_plan
/move_base/TebLocalPlannerROS/local_plan
/move_base/TebLocalPlannerROS/obstacles
/move_base/TebLocalPlannerROS/parameter_descriptions
/move_base/TebLocalPlannerROS/parameter_updates
/move_base/TebLocalPlannerROS/teb_feedback
/move_base/TebLocalPlannerROS/teb_markers
/move_base/TebLocalPlannerROS/teb_poses
/move_base/TebLocalPlannerROS/via_points
/move_base/cancel
/move_base/current_goal
/move_base/global_costmap/costmap
/move_base/global_costmap/costmap_prohibition_layer/parameter_descriptions
/move_base/global_costmap/costmap_prohibition_layer/parameter_updates
/move_base/global_costmap/costmap_updates
/move_base/global_costmap/footprint
/move_base/global_costmap/inflation_layer/parameter_descriptions
/move_base/global_costmap/inflation_layer/parameter_updates
/move_base/global_costmap/obstacle_layer/parameter_descriptions
/move_base/global_costmap/obstacle_layer/parameter_updates
/move_base/global_costmap/parameter_descriptions
/move_base/global_costmap/parameter_updates
/move_base/local_costmap/footprint
/move_base/local_costmap/inflation_layer/parameter_descriptions
/move_base/local_costmap/inflation_layer/parameter_updates
/move_base/local_costmap/obstacle_layer/parameter_descriptions
/move_base/local_costmap/obstacle_layer/parameter_updates
/move_base/local_costmap/parameter_descriptions
/move_base/local_costmap/parameter_updates
/move_base/local_costmap/sonar_layer1/parameter_descriptions
/move_base/local_costmap/sonar_layer1/parameter_updates
/move_base/local_costmap/static_layer/parameter_descriptions
/move_base/local_costmap/static_layer/parameter_updates
/move_base/parameter_descriptions
/move_base/parameter_updates
/move_base/result
/move_base/status
/move_base_simple/goal
/odom

i have this may topics and may more topics, i want total /move_base and /amcl to be shared with base station computer remaining should be ignored

atiderko, 2019-02-22:
You can specify these topics to sync with following parameter in launch file:
```
<node name="master_sync" pkg="master_sync_fkie" type="master_sync">
<rosparam param="sync_topics">[
    /all_ranges,
    /amcl/parameter_descriptions,
    /amcl/parameter_updates,
    /amcl_pose,
    ...
]</rosparam>
</node>
```
You should be careful to not run node with same name on base station.

atiderko, 2019-03-14:
I hope that works for you now. If you have further questions about this topic, you can reopen this issue!

closed 2019-03-14 by atiderko