fkie/multimaster_fkie #96: Only local master discovered

RCConnolly, 2019-02-22:
I am trying to set up a multimaster network using 2 Jackal UGV robots and a ground station computer, with each computer running it's own ROS (Indigo dist.) master and using Ubuntu 14.04. My issue is that when running the master_discovery command from either Jackal robot or the desktop, only the local ROS master node is discovered.

I am new to computer networking, but believe my issue is due to incorrect multicasting routing. When I ping 224.0.0.1 from one of the computers, only messages from that computer's IP are being received. One peculiarity I've noticed is that in the terminal output the following line is printed "Detected master discovery: http://localhost:11611", and in most other examples I have seen "localhost" is replaced with the machine's private address IP such as 192.xxx.xxx... or 10.xxx.xxx...

On each machine I've set the ROS_MASTER_URI to it's private address IP, and have also tried setting the ROS_HOSTNAME to the hostname associated with that IP and listed in the /etc/hosts file. I've attached the full terminal output when running the master_discovery command from the desktop. Any help would be much appreciated!

[master_discovery_debug.txt](https://github.com/fkie/multimaster_fkie/files/2892285/master_discovery_debug.txt)

atiderko, 2019-02-22:
If no explicit route for multicast traffic is set it is routed to default gateway. On a host with multiple connectors (Eth, WiFi) you can change the multicast routing by e.g. (temporary)
`sudo route add -net 224.0.0.0 netmask 224.0.0.0 eth0`
and check your routing table with `route -n`
Some routers (WiFi) does not forward multicast traffic. In this cases you can use `robot_hosts` parameter to specify hosts for unicast communication.

The ROS_MASTER_URI should contain the resolved __public__ machine name. You can use the public IP but ROS tries to resolve it to the machine name.

The ROS_HOSTNAME should be used carefully. If it is set the ROS-Nodes bind only to the given interface. Otherwise it uses all available interfaces.

In you log file I see two different names for your host `dukerama-Precision-WorkStation-T5500` and `ramaDesktop` (resolved by ROS). Can `ramaDesktop` be resolved by other hosts and reach (ping) the resolved address?

atiderko, 2019-03-14:
I hope it works now. If not, please reopen!

closed 2019-03-14 by atiderko