fkie/multimaster_fkie #98: Add --port tag to roscore in ROS tutorial

jellevos, 2019-03-14:
I saw a little inconsistency in the ROS tutorial:
http://wiki.ros.org/multimaster_fkie/Tutorials/Setup%20a%20ROS%20master%20synchronization

The `roscore` command does not have to abide to the `ROS_MASTER_URI` as mentioned in this issue (from the `ros_comm` repo):
https://github.com/ros/ros_comm/issues/907

It would be consistent to add a `--port 11311` tag to the first `roscore` command as well.

I am now busy setting this up for two machines. Is it any help if I write a small bit for the tutorial about setting it up for multiple machines?

atiderko, 2019-03-15:
Any help is welcome! For other users a tutorial for multiple hosts could be helpful.

You can also add the `--port 11311` while update the tutorial ;-)

jellevos, 2019-03-15:
Once I'm added to the ROS wiki whitelist I'll write something!

jellevos, 2019-03-18:
I have written a small part on it! Let me know if you require any changes.

atiderko, 2019-03-19:
it looks good to me!

Thank you very much for contributing!

If you have more tutorials you are welcome to share ;-)

closed 2019-03-19 by atiderko