fkie/multimaster_fkie #101: Can't recognize masters in VM

Afladimir1, 2019-04-29:
I followed http://digital.csic.es/bitstream/10261/133333/1/ROS-systems.pdf and I can echo the topic created in the VM but not the other way around. And now when I try to check the master inside the VM I get this:

`masters: 
  - 
    name: "ubuntu"
    uri: "http://ubuntu:11311/"
    timestamp: 1556551349.94
    timestamp_local: 1556551349.94
    online: True
    discoverer_name: "/master_discovery"
    monitoruri: "http://localhost:11611"
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 154, in run
    (client_sock, client_addr) = self.server_sock.accept()
  File "/usr/lib/python2.7/socket.py", line 206, in accept
    sock, addr = self._sock.accept()
  File "/usr/lib/python2.7/socket.py", line 174, in _dummy
    raise error(EBADF, 'Bad file descriptor')
error: [Errno 9] Bad file descriptor`

atiderko, 2019-04-30:
Are you sure the communication and hostname resolve between VM and outside works correct?
And ROS_MASTER_URI are set to the hostname and not to `localhost`?

Did you look into console output of the `master_discovery`? Are there some errors or warning?

Afladimir1, 2019-04-30:
First let me thank you for the fast answer.

Yes so the steps I did were the following:

- Added each hosts in /ect/hosts with their IP and hostname
- Run export ROS_MASTER_URI=http://IP:11312 with the IP of each host
- Checked the values of ip_forward and multicasting
- netstat -g and test with ping 224.0.0.1 also working
- Pinging each host worked as well
- Then I do the following commands in separate windows: roscore; rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1; rosrun master_sync_fkie master_sync;

In the Vms the error is the one I wrote in the 1st post, and on the main there are no errors.  
Can you provide an email so I can send you the print of all outputs from both host? Usernames and IPs a kinda sensitive information. Thank you

atiderko, 2019-04-30:
what happens if you call `rosservice call /rosout/get_loggers`?

Afladimir1, 2019-04-30:
Both show the same:

loggers: 
  - name: "ros"
    level: "info"
  - 
    name: "ros.roscpp"
    level: "info"
  - 
    name: "ros.roscpp.cached_parameters"
    level: "info"
  - 
    name: "ros.roscpp.roscpp_internal"
    level: "info"
  - 
    name: "ros.roscpp.roscpp_internal.connections"
    level: "info"
  - 
    name: "ros.roscpp.superdebug"
    level: "warn"

atiderko, 2019-04-30:
I have no Idea where the error comes from. I can't reproduce it in VirtualBox 6.0 with Ubuntu 18.04.
But I mean it is not the problem why you can't discover/see the topics in VM.

Try to check your Multicast traffic:
https://taosecurity.blogspot.com/2006/09/generating-multicast-traffic.html

or add `robot_hosts` parameter to `master_discovery`:
`rosrun master_discovery_fkie master_discovery _robot_hosts:=[VM_HOST_NAME]`

This adds explicit an unicast communication to given hosts.

Afladimir1, 2019-05-02:
Thank you! Adding robot_host in main worked, and now the VM recognizes it. The error from the first post still appears but it detects the other master.

closed 2019-05-02 by atiderko