fkie/multimaster_fkie #102: move_base node is not accepting relay topic

sunil1349, 2019-04-30:


atiderko, 2019-04-30:
did you run `master_discovery` on all robots and base station? And on base station `master_sync`? Which filter are you configured for `master_sync`? Is this topic or node filtered by your configuration?

Is the topic name of `relay` and `move_base` equal?

sunil1349, 2019-04-30:
yes i do run master_discovery on all robots and base station 
the thing is i used relay topic for amcl_pose in both robots is working 
but for move_base it is not working

atiderko, 2019-04-30:
Is /robot0/move_base a topic name or node name? I mean it should be a topic name.

sunil1349, 2019-04-30:
yes it is node name but i used relay for list of topics under move_base example.
`<rosparam param="sync_topics">['/map', 'robot0/move_base/goal', '/robot0/amcl_pose']</rosparam>`
it is not working for it even i given relay for particular topic

atiderko, 2019-04-30:
in your configuration only one move_base topic should be synchronized: `robot0/move_base/goal`. All other are ignored and on the base station not visible. But you should add `/` in the front of `robot0/move_base/goal`!

atiderko, 2019-06-25:
I hope this issue was solved, please reopen if not!

closed 2019-06-25 by atiderko