fkie/multimaster_fkie #103: losing connectivity

Gil401, 2019-05-02:
Hi,

I am running Multimaster on 2 devices with the following commands:
on 1:  

> rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
> rosrun master_sync_fkie master_sync __name:=station _ignore_nodes:=[/robot] 

on 2:

> rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
> rosrun master_sync_fkie master_sync __name:=robot _ignore_nodes:=[/station] 

at the begging it seems to work, but after a few seconds I do not see the topics / nodes of the other device.

when invoking 

> rorsservice call /master_discovery/list_masters

 I do see the other device.

Please assist.
Thanks

atiderko, 2019-05-02:
do you have nodes with same ROS name on robot and station?

Set the log level of master sync to DEBUG and see which nodes/topics are synchronized (set the parameter /master_sync/log_level to DEBUG to see what happens on start)

atiderko, 2019-05-02:
I see now, you rename the `master_sync` node. In this case you have to add ALL names on ALL hosts to the ignore list.
So it is better you do not rename `master_sync`!

Gil401, 2019-05-02:
so I need to simply run:

> rosrun master_discovery_fkie master_discovery _mcast_group:=224.0.0.1
> rosrun master_sync_fkie master_sync
> 

what if I have more robots with same nodes?

atiderko, 2019-05-02:
in ROS eco system you can not have two nodes with same name. So you need to launch your ROS nodes on each robot with a different namespace.

closed 2019-06-25 by atiderko