fkie/multimaster_fkie #105: Cannot hide topics between two identical masters

strid408, 2019-05-09:
Currently, we're trying to listen to just one topic from each master.
The two masters are identical and are both running the current parameters:

<rosparam param="sync_topics">[/truck_state0, /truck_state1]</rosparam>

Both masters are running a few nodes each, with their own published topics that are only supposed to be visible internally. However, the topics from each master seem to intertwine with the other. But that shouldn't be the case, because they are hidden when we use the sync topic parameter, right?

atiderko, 2019-05-10:
if you start with `master_sync` with `<rosparam param="sync_topics">[/truck_state0, /truck_state1]</rosparam>` only topics `/truck_state0` and `/truck_state1` are synchronized. It means if `master_sync` finds this topic on remote hosts he registers this topic **and** his node name on local master. So if you run locally a node with same name it will be stopped by ROS.

But `master_sync` does not sync  other topic. If so, this is not expected behaviour. In this case you should see into log output of `master_sync` (perhaps enable DEBUG level).

strid408, 2019-05-10:
> if you start with `master_sync` with `<rosparam param="sync_topics">[/truck_state0, /truck_state1]</rosparam>` only topics `/truck_state0` and `/truck_state1` are synchronized. It means if [...]

It seems like multimaster synchronized an rviz service by default so we simple added the below line which seemed to fix it..
`<rosparam param="ignore_services">['*']</rosparam>`

Thanks for the help!

closed 2019-05-10 by strid408