fkie/multimaster_fkie #107: Plans to release latest version as debian packages?

marco-tranzatto, 2019-06-10:
Hi @atiderko 

once again thanks for sharing this awesome tool!

I was wondering if there's any plan to release any time soon(-ish) the latest version of Master as debian packages, such that we can install it via "apt" (latest available version is  `0.8.12-0bionic.20190430.173238`). 

Cheers,
Marco.

atiderko, 2019-06-11:
Hi @marco-tranzatto 

I am glad you find this tool useful.

Unfortunately, I can not publish the latest version (with daemon instance) as a debian package through ros build system. This version uses python-grpcio  and python-grpc-tools which are not available for `bionic`.
The plan is to publish the new version with new ROS.

marco-tranzatto, 2019-06-12:
Hi @atiderko,

thanks for the clarification.
I asked because I have some conflicting issues with protobuf versions when I try to build your tool from source in the work-space I work with.
I was hoping to install it directly via apt and avoid some headaches ... :laughing: 

Cheers,
Marco.

closed 2019-06-13 by marco-tranzatto