fkie/multimaster_fkie #112: Problem when run multirobot simulation in Gazebo

ChunJyeBehBeh, 2019-09-03:
Hello, I would like to run multirobot simulation in Gazebo. The following is the set up.

PC 1: launch robot 1 and spawn it in Gazebo.
PC 2: launch robot 2 and spawn it in Gazebo(which on PC 1 via multimaster package).

However, the amcl of robot 2 will not launch properly which caused problems in the tf_tree.

I have tried to launch the two robots and Gazebo in PC 2 and there is no problem for it.

I have checked there is no topics in ignore_topics parameters. May I know how to troubleshooting or solve this problem?

atiderko, 2019-09-05:
Hi, did you check the time of both PCs? Tf tree allows only small tolerances.

Otherwise you can try launch 'node_manager' (GUI) to find the problem or provide your launch files.

ChunJyeBehBeh, 2019-09-05:
I have been checked the time of both PCs( which is GMT+8 , is this what you mean? or something about /clock?)

I launch node_manager and there is one warn in log
`[WARN] 05.09.2019 19:03:43.567681 (/node_manager_daemon:server.py:start:84): can not add insecure channel to '[::]:12321', try again...`
Is this the reason to cause multirobot simulation over gazebo fail to me?

Thanks for the reply.

atiderko, 2019-09-05:
I mean the time synchronization. If the time difference between hosts is more than 0.3-0.5 seconds you have to synchronize the time (with chrony or ntp).

The `node_manager_daemon` used only with `node_manager` and  not for synchronization. So it is not a reason for your problem. But the warning should not happen!

Can you provide the logs of `master_sync` and `master_discovery`? You can activate debug output with `rosrun fkie_master_sync master_sync _log_level:=DEBUG`. (Restart if no debug outputs are printed after first start.)

If you are able to start `node_manager` and you have time synchronization problems, you will see follow symbol in the GUI:
[image]

ChunJyeBehBeh, 2019-09-06:
Hello, I am able to start the GUI but I didn't see the symbol.

The following part is the debug message from the GUI.
```
[DEBUG] 06.09.2019 10:36:31.567737 (/node_manager:monitor_channel.py:get_system_diagnostics:67): get system diagnostics from grpc://beh-gl552vx:12321
[DEBUG] 06.09.2019 10:36:31.567737 (/node_manager:remote.py:get_insecure_channel:86): create insecure channel to beh-gl552vx:12321
[DEBUG] 06.09.2019 10:36:31.567737 (/node_manager:channel_interface.py:close_channel:61): close channel to beh-gl552vx:12321
```

It seem quite okay to me. Any other information that I can provide to you?

atiderko, 2019-09-06:
Can you provide the logs of `master_sync` and `master_discovery`? You can activate debug output with rosrun fkie_master_sync master_sync _log_level:=DEBUG. (Restart if no debug outputs are printed after first start.)

ChunJyeBehBeh, 2019-09-06:
But there is not debug outputs for me no matter how many times I restart the node.

atiderko, 2019-09-06:
did you start it with
```
fkie_master_sync master_sync _log_level:=DEBUG
```

ChunJyeBehBeh, 2019-09-06:
Yes, but there are only INFO message printed in the terminal.

atiderko, 2019-09-06:
:-/

If you provide me your launch files I can try to test they in my environment.

atiderko, 2019-09-10:
ok you can change
https://github.com/fkie/multimaster_fkie/blob/master/fkie_master_sync/src/fkie_master_sync/__init__.py#L83
and
https://github.com/fkie/multimaster_fkie/blob/master/fkie_master_discovery/src/fkie_master_discovery/__init__.py#L101
to `rospy.init_node(PROCESS_NAME, log_level=rospy.DEBUG)`

atiderko, 2019-12-17:
I hope, you solved your problem.

If not, please reopen!

closed 2019-12-17 by atiderko