fkie/multimaster_fkie #115: Unable to detect ROS master across Wifi and Ethernet

lyf44, 2019-10-29:
Hi, 

I have a PC connected to both Wifi and Ethernet. A ROS master is running in Wifi network A and another ROS master is running in Ethernet network B. At PC, I can ping both machines where ROS master is running. However, I am unable to establish communications between two ROS masters.

Currently, when I run the following in my laptop.
```
export ROS_MASTER_URI="http://10.66.171.1:11311" # ROS_MASTER_URI for ROS Master on network A
rosrun master_discovery_fkie master_discovery
```
I get
```
[INFO] [WallTime: 1572328934.230123] ROS Master URI: http://10.66.171.1:11311
[INFO] [WallTime: 1572328934.261932] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1572328934.262249] Heartbeat [Hz]: 0.02
[INFO] [WallTime: 1572328934.262452] Active request after [sec]: 60
[INFO] [WallTime: 1572328934.262631] Remove after [sec]: 300
[INFO] [WallTime: 1572328934.262809] Robot hosts: []
[INFO] [WallTime: 1572328934.262983] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1572328934.275997] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1572328934.276577] Subscribe to parameter /roslaunch/uris
[INFO] [WallTime: 1572328934.302547] + Bind to specified unicast socket @(137.132.82.147:11511)
[INFO] [WallTime: 1572328934.302908] Create multicast socket at ('226.0.0.0', 11511)
[INFO] [WallTime: 1572328934.624952] Detected master discovery: http://localhost:11611
[INFO] [WallTime: 1572328934.729224] Added master with ROS_MASTER_URI=http://10.66.171.1:11311/
```
And then I run the following in another terminal:
```
export ROS_MASTER_URI=http://137.132.82.147:11311 # ROS_MASTER_URI on network B
rosrun master_discovery_fkie master_discovery
```
I get:
```
[INFO] [WallTime: 1572329338.024654] ROS Master URI: http://localhost:11311
[INFO] [WallTime: 1572329338.041032] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1572329338.041357] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1572329338.041591] Active request after [sec]: 60
[INFO] [WallTime: 1572329338.041808] Remove after [sec]: 300
[INFO] [WallTime: 1572329338.042024] Robot hosts: []
[INFO] [WallTime: 1572329338.042234] Approx. mininum avg. network load: 1.36 bytes/s
[WARN] [WallTime: 1572329338.050148] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use Try again...
[WARN] [WallTime: 1572329339.051930] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use Try again...
[WARN] [WallTime: 1572329340.053891] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use Try again...
[WARN] [WallTime: 1572329341.055901] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use Try again...
[WARN] [WallTime: 1572329342.058117] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use Try again...
```

My current multicast setting and ip forward setting is as follows:
```cat /proc/sys/net/ipv4/ip_forward -> 1 # ip forward is enabled```
```cat /proc/sys/net/ipv4/icmp_echo_ignore_broadcasts -> 1 # multi cast not enabled```

I deeply appreciate if you can help with this issue. Thank you a lot in advance.

atiderko, 2019-10-29:
Hi @lyf44,

on second computer the `master_discovery` is not started correctly, because the port 11611 is blocked. Use another mcast_port for all `master_discovery` instances to change the RPC-XML port, e.g.
`rosrun master_discovery_fkie master_discovery _mcast_port:=11512`

By default all multicast messages are routed to default route. On PC with multiple network interfaces it can be the wrong interface. You can list your routes with `route -n`.
You can try to change the multicast route with e.g. 
`sudo route add -net 224.0.0.0 netmask 224.0.0.0 eth0`

Or you configure `robot_hosts`. In this case unicast messages are send to defined hosts.
@10.66.171.1:
`rosrun master_discovery_fkie master_discovery _robot_hosts:=[137.132.82.147] _mcast_port:=11512`
and @137.132.82.147:
`rosrun master_discovery_fkie master_discovery _robot_hosts:=[10.66.171.1] _mcast_port:=11512`

PS: you can use `node_manager` to start discovery nodes and control your ROS nodes.

lyf44, 2019-10-30:
Hi atiderko,

Thank you for your reply. Unfortunately it does not resolve the issue for me. I am currently running master discovery **in two terminals inside the same machine** as mentioned in my previous posts and running
```rosrun master_discovery_fkie master_discovery _robot_hosts:=[137.132.82.147] _mcast_port:=11512```
in one terminal and running  
```rosrun master_discovery_fkie master_discovery _robot_hosts:=[10.66.171.1] _mcast_port:=11512```
in another terminal still yields the same warning.
```[WARN] [WallTime: 1572329338.050148] Error while start RPC-XML server on port 11611: [Errno 98] Address already in use Try again...```

Unfortunately I am not very familiar with route settings. The following is my route -n setting:
```
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         137.132.82.1    0.0.0.0         UG    0      0        0 eth0
10.66.171.0     0.0.0.0         255.255.255.0   U     9      0        0 wlan0
137.132.82.0    0.0.0.0         255.255.255.0   U     1      0        0 eth0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
```
Do I have to run master discovery on different machines? If not, do I need to modify the route settings to make it work?

atiderko, 2019-10-30:
Hi lyf44,

sorry, I misunderstood your scenario in first post.

The `master_discovery` is designed to use on the same machine as ROS Master. There is a lot of communication between ROS Master and `master_discovery` so you should be careful if you start it on remote host.

If you want to do it anyway, try to change the `rpc_port` on second `master_discovery`, e.g.:

```
rosrun master_discovery_fkie master_discovery _rpc_port:=11612
```

lyf44, 2019-10-30:
Hi atiderko,

Thank you for your reply. Now I am able to run 
```rosrun master_discovery_fkie master_discovery _rpc_port:=11612``` 
in one terminal and
```rosrun master_discovery_fkie master_discovery``` in the other.

The first terminal gives:
```
[INFO] [WallTime: 1572425352.909207] ROS Master URI: http://137.132.82.147:11311
[INFO] [WallTime: 1572425352.924244] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1572425352.924597] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1572425352.924842] Active request after [sec]: 60
[INFO] [WallTime: 1572425352.925056] Remove after [sec]: 300
[INFO] [WallTime: 1572425352.925262] Robot hosts: []
[INFO] [WallTime: 1572425352.925437] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1572425352.933093] Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [WallTime: 1572425352.933659] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1572425352.940159] + Bind to specified unicast socket @(137.132.82.147:11511)
[INFO] [WallTime: 1572425352.940505] Create multicast socket at ('226.0.0.0', 11511)
[INFO] [WallTime: 1572425353.162378] Detected master discovery: http://localhost:11611
[INFO] [WallTime: 1572425353.266388] Added master with ROS_MASTER_URI=http://137.132.82.147:11311/
[INFO] [WallTime: 1572425353.478301] Detected master discovery: http://localhost:11612
[INFO] [WallTime: 1572425353.581855] Added master with ROS_MASTER_URI=http://10.66.171.1:11311/
[WARN] [WallTime: 1572425354.059871] Resolved host of ROS_MASTER_URI 10.66.171.1=10.66.171.1 and origin discovered IP=localhost are different. Fix your network settings and restart master_discovery!
```
and the second terminal gives:
```
[INFO] [WallTime: 1572425526.621264] ROS Master URI: http://10.66.171.1:11311
[INFO] [WallTime: 1572425526.655630] Check the ROS Master[Hz]: 1
[INFO] [WallTime: 1572425526.655932] Heart beat [Hz]: 0.02
[INFO] [WallTime: 1572425526.656131] Active request after [sec]: 60
[INFO] [WallTime: 1572425526.656306] Remove after [sec]: 300
[INFO] [WallTime: 1572425526.656479] Robot hosts: []
[INFO] [WallTime: 1572425526.656662] Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [WallTime: 1572425526.686432] Start RPC-XML Server at ('0.0.0.0', 11612)
[INFO] [WallTime: 1572425526.689205] Subscribe to parameter `/roslaunch/uris`
[INFO] [WallTime: 1572425526.714735] + Bind to specified unicast socket @(137.132.82.147:11511)
[INFO] [WallTime: 1572425526.715122] Create multicast socket at ('226.0.0.0', 11511)
[INFO] [WallTime: 1572425527.037455] Detected master discovery: http://localhost:11612
[INFO] [WallTime: 1572425527.140667] Added master with ROS_MASTER_URI=http://10.66.171.1:11311/
[INFO] [WallTime: 1572425527.770953] Detected master discovery: http://localhost:11611
[INFO] [WallTime: 1572425527.874091] Added master with ROS_MASTER_URI=http://137.132.82.147:11311/
```
It seems the master discovery works now despite the warning in the first terminal output. Then I open two more terminals and run ```rosrun master_sync_fkie master_sync``` in both of them, one runs properly but another gives the following:
```
[INFO] [WallTime: 1572425927.485012] ignore_hosts: []
[INFO] [WallTime: 1572425927.487609] sync_hosts: []
[INFO] [WallTime: 1572425927.490500] sync_topics_on_demand: False
[INFO] [WallTime: 1572425927.494910] resync_on_reconnect: True
[INFO] [WallTime: 1572425927.497831] resync_on_reconnect_timeout: 0
[WARN] [WallTime: 1572425927.536586] Master_discovery node appear not to running. Wait for topic with type 'MasterState.
[WARN] [WallTime: 1572425928.565478] Master_discovery node appear not to running. Wait for topic with type 'MasterState.
[WARN] [WallTime: 1572425929.604235] Master_discovery node appear not to running. Wait for topic with type 'MasterState.
[WARN] [WallTime: 1572425930.645957] Master_discovery node appear not to running. Wait for topic with type 'MasterState.
```
Could you help me with the problem? Thank you so much for the help.

atiderko, 2019-10-30:
In this case, the solution is a little more complicated.

Since the `master_sync` is not running on the same host with ROS Master the check for hostname fails. It was already addressed in issue #106. There is a solution for this: you can set `check_host` to `false` while start `master_sync`.

But you have to use the current github version of multimaster. In the current version all packages are renamed from *_fkie to fkie_*.

lyf44, 2019-10-31:
Hi atiderko,

Thank you very much for your reply. I have downloaded the current version and run ```rosrun master_sync_fkie master_sync _check_host:=false``` and everything runs smoothly now.

One last question, do I have to take care of this warning? If yes, how?
```[WARN] [WallTime: 1572425354.059871] Resolved host of ROS_MASTER_URI 10.66.171.1=10.66.171.1 and origin discovered IP=localhost are different. Fix your network settings and restart master_discovery!```

Thanks a lot for the help so far. You can close this issue now.

atiderko, 2019-10-31:
Nice to hear it works for you!

In usual case the `master_discovery` and ROS Master are running at the same host. So they address should be equal.
Your scenario is different from usual case. So if your ROS communication works well you can ignore this warning.

closed 2019-11-01 by lyf44