fkie/multimaster_fkie #117: Screen output on node_manager not working

weixuanzhang, 2019-12-17:
Hi,
I recently updated the node_manager, I am not able to output node to the terminal anymore. 
A snippet from /var/log/dpkg.log
2019-12-02 17:14:24 upgrade ros-kinetic-node-manager-fkie:amd64 0.8.11-0xenial-20190227-103017-0800 0.8.12-0xenial-20191029-002604+0000

When I click the output button on node_manager, the terminal that runs node_manager outputs:
[INFO] [1576578161.392919]: Open screen terminal: /usr/bin/x-terminal-emulator -T "SCREEN /omav/omav_allocation_node on weixuan-blade"  -e /usr/bin/screen -x 6913._omav_omav_allocation_node

But no external screen is spawn. 
My system setup:
System: Ubuntu 16.04,
ROS: kinetic
Terminal: terminator 1.91

When I run "sudo update-alternatives --config x-terminal-emulator"
I choose
 Selection    Path                             Priority   Status
------------------------------------------------------------
* 0            /usr/bin/terminator.wrapper       50        auto mode

Any help much appreciated! Thanks!

atiderko, 2019-12-17:
Hi,

can you please check two things (in separate terminal):
1. Is the screen running: `screen -ls`
2. Does screen open if you run: `/usr/bin/x-terminal-emulator -T "SCREEN /omav/omav_allocation_node on weixuan-blade" -e /usr/bin/screen -x 6913._omav_omav_allocation_node` <-- This is the line which node_manager prints and executes to open screen terminal

weixuanzhang, 2019-12-17:
Thank you very much for your quick response!
1. When I run screen -ls:
 There are screens on:
	7369._omav_spawn_omav	(17.12.2019 11:22:36)	(Detached)
	7081._omav_omav_controller_node	(17.12.2019 11:22:36)	(Detached)
	6913._omav_omav_allocation_node	(17.12.2019 11:22:35)	(Detached)
	6832._omav_local_planner	(17.12.2019 11:22:35)	(Detached)
	6727._omav_full_pose_waypoint_node	(17.12.2019 11:22:35)	(Detached)
	3654._roscore--11311	(17.12.2019 11:14:50)	(Detached)
They are the running nodes in the node manager.
6 Sockets in /var/run/screen/S-weixuan.

2.When I run  /usr/bin/x-terminal-emulator -T "SCREEN /omav/omav_allocation_node on weixuan-blade" -e /usr/bin/screen -x 6913._omav_omav_allocation_node
I get
terminator: error: Additional unexpected arguments found: ['/omav/omav_allocation_node', 'on', 'weixuan-blade']

Thanks!

atiderko, 2019-12-17:
I tested it with terminator v1.91 on Ubuntu 18.04 and the command line works.

I think it is something with selected alternatives for x-terminal-emulator. Can you change it to terminator?

weixuanzhang, 2020-01-02:
Thank you very much for your input! Indeed it has something to do with the selected alternatives for x-terminal-emulator. The catch is that I installed the terminator 1.91 from ppa:gnome-terminator/nightly-gtk3. When setting default terminal, I have to switch from /usr/bin/terminator.wrapper to  /usr/bin/gnome-terminal.wrapper. I will close this issue.

closed 2020-01-02 by weixuanzhang