fkie/multimaster_fkie #119: Not all topics synched

NeelzDev, 2020-01-27:
Hello,

I am currently working on a Kubernetes cluster using ROS melodic (for specific project reasons melodic needs to be used). As the ros master is a single point of failure, I want to make it redundant and came across your software. Thanks for sharing it!

In the cluster every node can directly address every node via its IP.

Each ros master is running in a Kubernetes pod, this means each container inside the pod shares storage and network. In total there are three containers for each pod running the following commands:

1. roscore: `. /opt/ros/melodic/setup.bash; export ROS_IP="$(hostname -i)"; roscore`
2. master discovery: `. /opt/ros/melodic/setup.bash; export ROS_IP="$(hostname -i)"; rosrun master_discovery_fkie master_discovery`
3. master sync: `. /opt/ros/melodic/setup.bash; export ROS_IP="$(hostname -i)"; rosrun master_sync_fkie master_sync`

The container image is created as following:
```
FROM ros:melodic-ros-base
RUN apt-get update && apt-get install -y ros-melodic-multimaster-fkie
```
Note: Broadcast and multicast get enabled before start.

The master discovery seems to work correctly:
Pod 1  (master discovery output):
```
[INFO] [1580121414.848866]: Check the ROS Master[Hz]: 1
[INFO] [1580121414.849375]: Heart beat [Hz]: 0.02
[INFO] [1580121414.849813]: Active request after [sec]: 60
[INFO] [1580121414.850235]: Remove after [sec]: 300
[INFO] [1580121414.850663]: Robot hosts: []
[INFO] [1580121414.851079]: Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [1580121414.855287]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1580121414.855983]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1580121414.858357]: + Bind to specified unicast socket @(10.42.0.129:11511)
[INFO] [1580121414.858861]: Create multicast socket at ('226.0.0.0', 11511)
[INFO] [1580121415.027750]: Detected master discovery: http://localhost:11611
[INFO] [1580121415.131294]: Added master with ROS_MASTER_URI=http://10.42.0.129:11311/
[INFO] [1580121415.443574]: Detected master discovery: http://10.42.0.130:11611
[INFO] [1580121415.547302]: Added master with ROS_MASTER_URI=http://10.42.0.130:11311/
```

Pod 2  (master discovery output):
```
[INFO] [1580121414.370170]: ROS Master URI: http://localhost:11311
[INFO] [1580121414.375819]: Check the ROS Master[Hz]: 1
[INFO] [1580121414.376521]: Heart beat [Hz]: 0.02
[INFO] [1580121414.377118]: Active request after [sec]: 60
[INFO] [1580121414.377603]: Remove after [sec]: 300
[INFO] [1580121414.378058]: Robot hosts: []
[INFO] [1580121414.378539]: Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [1580121414.382736]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1580121414.383488]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1580121414.386952]: + Bind to specified unicast socket @(10.42.0.130:11511)
[INFO] [1580121414.387578]: Create multicast socket at ('226.0.0.0', 11511)
[INFO] [1580121414.560276]: Detected master discovery: http://localhost:11611
[INFO] [1580121414.662968]: Added master with ROS_MASTER_URI=http://10.42.0.130:11311/
[INFO] [1580121415.025819]: Detected master discovery: http://10.42.0.129:11611
[INFO] [1580121415.128332]: Added master with ROS_MASTER_URI=http://10.42.0.129:11311/
```

Also I do not see any errors in the sync output:

Pod 1 (master sync output):
```
...
[INFO] [1580121446.300150]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.129', '10.42.0.130']
[INFO] [1580121470.236891]: SyncThread[10.42.0.129] Requesting remote state from 'http://10.42.0.129:11611'
[INFO] [1580121470.240626]: SyncThread[10.42.0.129] Applying remote state...
[INFO] [1580121470.241990]: SyncThread[10.42.0.129] remote state applied.
[INFO] [1580121472.766345]: SyncThread[10.42.0.129] Requesting remote state from 'http://10.42.0.129:11611'
[INFO] [1580121472.770274]: SyncThread[10.42.0.129] Applying remote state...
[INFO] [1580121472.773116]: SyncThread[10.42.0.129] remote state applied.
[INFO] [1580121476.307809]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.129', '10.42.0.130']
[INFO] [1580121506.315516]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.129', '10.42.0.130']
[INFO] [1580121536.322906]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.129', '10.42.0.130']
[INFO] [1580121566.329678]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.129', '10.42.0.130']
...
```

Pod 2 (master sync output):
```
...
[INFO] [1580123187.126108]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.130', '10.42.0.129']
[INFO] [1580123208.181042]: SyncThread[10.42.0.130] Requesting remote state from 'http://10.42.0.130:11611'
[INFO] [1580123208.185128]: SyncThread[10.42.0.130] Applying remote state...
[INFO] [1580123208.187537]: SyncThread[10.42.0.130] remote state applied.
[INFO] [1580123210.238873]: SyncThread[10.42.0.130] Requesting remote state from 'http://10.42.0.130:11611'
[INFO] [1580123210.242788]: SyncThread[10.42.0.130] Applying remote state...
[INFO] [1580123210.245158]: SyncThread[10.42.0.130] remote state applied.
[INFO] [1580123217.137040]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.130', '10.42.0.129']
[INFO] [1580123217.641776]: SyncThread[10.42.0.130] Requesting remote state from 'http://10.42.0.130:11611'
[INFO] [1580123217.645935]: SyncThread[10.42.0.130] Applying remote state...
[INFO] [1580123217.648215]: SyncThread[10.42.0.130] remote state applied.
[INFO] [1580123247.147505]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.130', '10.42.0.129']
[INFO] [1580123277.158128]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.130', '10.42.0.129']
...
```

For test purpose there is realsense (camera) node which is publishing data from the camera. This will create a certain amount of topics and the realsense randomly registers add one of the masters. 

At master one, I see the following topics:
```
/camera/accel/imu_info
/camera/accel/sample
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/aligned_depth_to_infra1/camera_info
/camera/aligned_depth_to_infra1/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_rect_color
/camera/color_rectify_color/parameter_descriptions
/camera/color_rectify_color/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth_registered/points
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_infra1
/camera/extrinsics/depth_to_infra2
/camera/gyro/imu_info
/camera/gyro/sample
/camera/infra1/camera_info
/camera/infra1/image_rect_raw
/camera/infra2/camera_info
/camera/infra2/image_rect_raw
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/master_discovery/changes
/master_discovery/linkstats
/rosout
/rosout_agg
/tf
/tf_static
```

At master 2 there is only a small subset of the topics available:
```
/camera/color/image_rect_color
/camera/color_rectify_color/parameter_descriptions
/camera/color_rectify_color/parameter_updates
/camera/realsense2_camera_manager/bond
/master_discovery/changes
/master_discovery/linkstats
/rosout
/rosout_agg
```

It looks like there seems to be some synchronization but it seems to not work completely. 

Do you see any problems in the set up? Is there some output/log files I should consult?

Thank you in advance!

Nils

atiderko, 2020-01-27:
Hi,

is there a reason why you use `export ROS_IP="$(hostname -i)";`?
Did you also use `export ROS_IP="$(hostname -i)";` while you start realsense node?

It could be a problem if ROS_MASTER_URI and ROS_IP has not the same entry, since they are not resolved by master sync.

For more output you can run master_sync with `_log_level:=DEBUG`
```
rosrun fkie_master_sync master_sync _log_level:=DEBUG
```
(run twice, if you do no see debug output)

NeelzDev, 2020-01-27:
Thanks for the quick reply!

The `export ROS_IP="$(hostname -i)";` is done, because the hostname is per default not resolvable in kuberenetes. I did so to achieve that the IP address is used instead of the hostname. Which seems to work if I check the discover output:
`[INFO] [1580121414.662968]: Added master with ROS_MASTER_URI=http://10.42.0.130:11311/``

For the realsense this is not done . Instead its ROS_HOSTNAME is set to the name (in this case "realsense") of a "headless kubernetes service" which is pointing to the pod and the container contained. 

With just one master this is done also for the ros master. But with multiple masters the service "ros-master" should resolve to all instances and connection will be randomly distributed between the masters. Following this, it should be transparent for a client if a master goes down or comes up, if all masters are synched.
Therefor, I used the trick of setting the ROS_IP in order to not use the ROS_HOSTNAME.

Just as a side note: With just one standard ros master everything is working as expected. 

I tested with setting the ROS_MASTER_URI to `export ROS_IP="http://$(hostname -i):11311"`. Unfortunately this did not solve the problem.

I will try with investigating the logs in more details this evening with the additional log level!

At which frequency are the masters synchronized?

atiderko, 2020-01-28:
Too bad it didn't work.

Did you set 
`export ROS_MASTER_URI=http://$(hostname -i):11311`
and
`export ROS_IP=$(hostname -i)`
for all ROS-Nodes (roscore, master_discovery, master_sync and realsense)?

By default the changes of local roscore are checked at 1 Hz. If changes are detected the synchronization process will be initialized. You can change the check rate in `master_discovery` by `rosmaster_hz` parameter.

NeelzDev, 2020-01-30:
Thanks for your support!

So I switched to the master branch by basically changing the docker image to the following:
```
FROM ros:melodic-ros-base

RUN apt-get update && apt-get install -y \
    python-pip python-catkin-tools

RUN pip install grpcio-tools grpcio

RUN mkdir -p catkin_ws/src

WORKDIR 'catkin_ws/src'
RUN git clone https://github.com/fkie/multimaster_fkie.git multimaster && \ 
    rosdep update && \
    rosdep install -i --as-root pip:false --reinstall --from-paths multimaster -y 

WORKDIR '..'

RUN /bin/bash -c '. /opt/ros/melodic/setup.bash; catkin build fkie_multimaster'
```

This works fine and the software is running. Unfortunately the issue is not solved.

In the debug log output I see that the sync thread is receiving the correct set of remote topics:
```
[INFO] [1580395456.222212]: SyncThread[10.42.0.67] Requesting remote state from 'http://10.42.0.67:11611'
[INFO] [1580395456.227957]: SyncThread[10.42.0.67] Applying remote state...
[DEBUG] [1580395456.228905]: SyncThread[10.42.0.67]: remote state: ['1580395436.622139931', '1580395436.622139931', 'http://10.42.0.67:11311/', '10.42.0.67', [['/camera/rgb_camera/parameter_descriptions', ['/camera/realsense2_camera_manager']], ['/camera/depth_registered/points', ['/camera/realsense2_camera_manager']], ['/camera/color_rectify_color/parameter_descriptions', ['/camera/realsense2_camera_manager']], 
...

[DEBUG] [1580395456.230676]: SyncThread[10.42.0.67]: current timestamp 1580395436.622139931, local 1580395436.622139931
[INFO] [1580395456.232133]: SyncThread[10.42.0.67] Requesting remote md5sums 'http://10.42.0.67:11611'
[INFO] [1580395456.233200]: SyncThread[10.42.0.67] remote state applied.
```

Following this, I think the issue might be that the remote state is actually not applied but no error is raised.
Do you have any idea what could be the reason or what I could investigate?

atiderko, 2020-01-30:
Thank you for switching the multimaster version!

Can you post the whole debug output?

I do not see any subscriptions. So the node or topics are filtered for some reason. It would be interesting to see the complete remote state.

NeelzDev, 2020-01-30:
Master 1 (discovery):
```
[DEBUG] [1580397599.077584]: init_node, name[/master_discovery], pid[20769]
[DEBUG] [1580397599.080289]: binding to 0.0.0.0 0
[DEBUG] [1580397599.082714]: bound to 0.0.0.0 35883
[DEBUG] [1580397599.085467]: ... service URL is rosrpc://10.42.0.67:35883
[DEBUG] [1580397599.087769]: [/master_discovery/get_loggers]: new Service instance
[DEBUG] [1580397599.092899]: ... service URL is rosrpc://10.42.0.67:35883
[DEBUG] [1580397599.094506]: [/master_discovery/set_logger_level]: new Service instance
[INFO] [1580397599.103507]: ROS Master URI: http://10.42.0.67:11311
[INFO] [1580397599.109832]: Check the ROS Master[Hz]: 1
[INFO] [1580397599.110531]: Heart beat [Hz]: 0.02
[INFO] [1580397599.111186]: Active request after [sec]: 60
[INFO] [1580397599.111882]: Remove after [sec]: 300
[INFO] [1580397599.112760]: Robot hosts: []
[INFO] [1580397599.113401]: Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [1580397599.117416]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1580397599.118155]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1580397599.120564]: + Bind to specified unicast socket @(10.42.0.67:11511)
[DEBUG] [1580397599.121114]: Ucast bind to: (10.42.0.67:11511)
[DEBUG] [1580397599.121615]: mgroup: 226.0.0.0
[DEBUG] [1580397599.122106]: interface : 10.42.0.67
[DEBUG] [1580397599.122585]: inet: (2, 1, 6, '', ('10.42.0.67', 0))
[INFO] [1580397599.123066]: Create multicast socket at ('226.0.0.0', 11511)
[DEBUG] [1580397599.124480]: node[/master_discovery, http://10.42.0.67:44421/] entering spin(), pid[20769]
[DEBUG] [1580397599.280103]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397599.292285]: Received a NEW heartbeat from ('localhost', 11511) via MULTICAST socket
[INFO] [1580397599.293136]: Detected master discovery: http://localhost:11611
[DEBUG] [1580397599.394004]: Get additional connection info from http://localhost:11611
[DEBUG] [1580397599.397767]: Got [1580397599.224494934, http://10.42.0.67:11311/, 10.42.0.67, /master_discovery] from http://10.42.0.67:11611/
[INFO] [1580397599.399907]: Added master with ROS_MASTER_URI=http://10.42.0.67:11311/
[DEBUG] [1580397599.401989]: ... service URL is rosrpc://10.42.0.67:35883
[DEBUG] [1580397599.403590]: [/master_discovery/list_masters]: new Service instance
[DEBUG] [1580397599.407248]: ... service URL is rosrpc://10.42.0.67:35883
[DEBUG] [1580397599.409161]: [/master_discovery/refresh]: new Service instance
[DEBUG] [1580397600.125092]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397600.126744]: Send requests while init 1/3
[DEBUG] [1580397600.128040]: Send request to mcast group 226.0.0.0:11511
[DEBUG] [1580397600.129465]: Set timer to send heartbeat in 50.00 sec
[DEBUG] [1580397600.158603]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397600.160119]: Received a multicast request for a state update from localhost
[DEBUG] [1580397600.161519]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397600.170077]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397600.303342]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397600.335697]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397612.802829]: Received a NEW heartbeat from ('10.42.0.64', 11511) via MULTICAST socket
[INFO] [1580397612.804244]: Detected master discovery: http://10.42.0.64:11611
[DEBUG] [1580397612.905910]: Get additional connection info from http://10.42.0.64:11611
[DEBUG] [1580397612.909096]: Got [1580397612.698462009, http://10.42.0.64:11311/, 10.42.0.64, /master_discovery] from http://10.42.0.64:11611/
[INFO] [1580397612.910470]: Added master with ROS_MASTER_URI=http://10.42.0.64:11311/
[DEBUG] [1580397613.670592]: Received a heartbeat from ('10.42.0.64', 11511) via MULTICAST socket
[DEBUG] [1580397613.672545]: Received a multicast request for a state update from 10.42.0.64
[DEBUG] [1580397613.674588]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397613.676821]: Received a heartbeat from ('10.42.0.64', 11511) via MULTICAST socket
[DEBUG] [1580397613.710313]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397613.775292]: Received a heartbeat from ('10.42.0.64', 11511) via MULTICAST socket
[DEBUG] [1580397617.689237]: connecting to 10.42.0.67 39881
[DEBUG] [1580397617.691790]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397617.741956]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397620.957499]: Received a heartbeat from ('10.42.0.64', 11511) via MULTICAST socket
[DEBUG] [1580397650.131078]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397650.132437]: Send requests while init 2/3
[DEBUG] [1580397650.133514]: Send request to mcast group 226.0.0.0:11511
[DEBUG] [1580397650.134536]: Set timer to send heartbeat in 50.00 sec
[DEBUG] [1580397650.176232]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397650.177576]: Received a multicast request for a state update from localhost
[DEBUG] [1580397650.178693]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580397650.183048]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580397650.215643]: Received a heartbeat from ('10.42.0.64', 11511) via MULTICAST socket
```
[... 150 lines cut ...]
[INFO] [1580397620.439022]: [10.42.0.67] ignore_services: [u'/*get_loggers', u'/*set_logger_level']
[... cut ...]
[DEBUG] [1580397680.473596]: connecting to 10.42.0.64 41389
[INFO] [1580397680.477427]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.67', '10.42.0.64']
[DEBUG] [1580397710.485433]: connecting to 10.42.0.64 41389
[INFO] [1580397710.487775]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.67', '10.42.0.64']
[DEBUG] [1580397740.496448]: connecting to 10.42.0.64 41389
[INFO] [1580397740.498627]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.67', '10.42.0.64']
```

NeelzDev, 2020-01-30:
Simplified example with just one topic /test:

Master 1 (discovery):
```
[DEBUG] [1580398374.212741]: init_node, name[/master_discovery], pid[424]
[DEBUG] [1580398374.214056]: binding to 0.0.0.0 0
[DEBUG] [1580398374.215250]: bound to 0.0.0.0 44063
[DEBUG] [1580398374.216613]: ... service URL is rosrpc://10.42.0.72:44063
[DEBUG] [1580398374.217959]: [/master_discovery/get_loggers]: new Service instance
[DEBUG] [1580398374.220859]: ... service URL is rosrpc://10.42.0.72:44063
[DEBUG] [1580398374.222249]: [/master_discovery/set_logger_level]: new Service instance
[INFO] [1580398374.231701]: ROS Master URI: http://localhost:11311
[INFO] [1580398374.241155]: Check the ROS Master[Hz]: 1
[INFO] [1580398374.242277]: Heart beat [Hz]: 0.02
[INFO] [1580398374.243334]: Active request after [sec]: 60
[INFO] [1580398374.244432]: Remove after [sec]: 300
[INFO] [1580398374.245511]: Robot hosts: []
[INFO] [1580398374.246505]: Approx. mininum avg. network load: 1.36 bytes/s
[INFO] [1580398374.254377]: Start RPC-XML Server at ('0.0.0.0', 11611)
[INFO] [1580398374.255901]: Subscribe to parameter `/roslaunch/uris`
[INFO] [1580398374.260759]: + Bind to specified unicast socket @(10.42.0.72:11511)
[DEBUG] [1580398374.261891]: Ucast bind to: (10.42.0.72:11511)
[DEBUG] [1580398374.263058]: mgroup: 226.0.0.0
[DEBUG] [1580398374.264157]: interface : 10.42.0.72
[DEBUG] [1580398374.265254]: inet: (2, 1, 6, '', ('10.42.0.72', 0))
[INFO] [1580398374.266339]: Create multicast socket at ('226.0.0.0', 11511)
[DEBUG] [1580398374.269462]: node[/master_discovery, http://10.42.0.72:43975/] entering spin(), pid[424]
[DEBUG] [1580398374.414926]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398374.437285]: Received a NEW heartbeat from ('localhost', 11511) via MULTICAST socket
[INFO] [1580398374.438618]: Detected master discovery: http://localhost:11611
[DEBUG] [1580398374.539770]: Get additional connection info from http://localhost:11611
[DEBUG] [1580398374.543673]: Got [1580398374.369251966, http://10.42.0.72:11311/, 10.42.0.72, /master_discovery] from http://10.42.0.72:11611/
[INFO] [1580398374.545213]: Added master with ROS_MASTER_URI=http://10.42.0.72:11311/
[DEBUG] [1580398374.546637]: ... service URL is rosrpc://10.42.0.72:44063
[DEBUG] [1580398374.547967]: [/master_discovery/list_masters]: new Service instance
[DEBUG] [1580398374.551022]: ... service URL is rosrpc://10.42.0.72:44063
[DEBUG] [1580398374.552378]: [/master_discovery/refresh]: new Service instance
[DEBUG] [1580398375.270085]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398375.272297]: Send requests while init 1/3
[DEBUG] [1580398375.273775]: Send request to mcast group 226.0.0.0:11511
[DEBUG] [1580398375.275237]: Set timer to send heartbeat in 50.00 sec
[DEBUG] [1580398375.304319]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580398375.306244]: Received a multicast request for a state update from localhost
[DEBUG] [1580398375.307903]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398375.313231]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580398375.420050]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398375.428419]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580398375.943530]: Received a NEW heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[INFO] [1580398375.944951]: Detected master discovery: http://10.42.0.71:11611
[DEBUG] [1580398376.046312]: Get additional connection info from http://10.42.0.71:11611
[DEBUG] [1580398376.048980]: Got [1580398375.837663889, http://10.42.0.71:11311/, 10.42.0.71, /master_discovery] from http://10.42.0.71:11611/
[INFO] [1580398376.050087]: Added master with ROS_MASTER_URI=http://10.42.0.71:11311/
[DEBUG] [1580398376.810882]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398376.813009]: Received a multicast request for a state update from 10.42.0.71
[DEBUG] [1580398376.814966]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398376.816645]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398376.849588]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580398376.914780]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398384.556183]: connecting to 10.42.0.72 46587
[DEBUG] [1580398384.559525]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398384.631703]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[DEBUG] [1580398388.098189]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398391.131749]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398392.164459]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398395.229290]: Received a heartbeat from ('10.42.0.71', 11511) via MULTICAST socket
[DEBUG] [1580398425.277383]: Send current state to group 226.0.0.0:11511
[DEBUG] [1580398425.278970]: Send requests while init 2/3
[DEBUG] [1580398425.280424]: Send request to mcast group 226.0.0.0:11511
[DEBUG] [1580398425.281850]: Set timer to send heartbeat in 50.00 sec
[DEBUG] [1580398425.347815]: Received a heartbeat from ('localhost', 11511) via MULTICAST socket
[... 258 lines cut ...]
[DEBUG] [1580398396.675151]: SyncThread[10.42.0.72]: current timestamp 1580398384.540823936, local 1580398384.540823936
[... cut ...]
[DEBUG] [1580398545.013630]: connecting to 10.42.0.71 42369
[INFO] [1580398545.016006]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.72', '10.42.0.71']
[DEBUG] [1580398575.023781]: connecting to 10.42.0.71 42369
[INFO] [1580398575.025973]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.72', '10.42.0.71']
[DEBUG] [1580398605.034208]: connecting to 10.42.0.71 42369
[INFO] [1580398605.036965]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.72', '10.42.0.71']
```

atiderko, 2020-01-31:
I think, the problem is the different ROS_MASTER_URI ('http://10.42.0.71:11311/') and Node RPC URI ('http://basic-pipeline-nils-ros-vnc:39653/'). Since the check is only done by comparing the host names of these two URLs, the nodes are not synchronized.

I have added a fix that also resolves the hostname and IP addresses. I published it to new branch `fix_sync_ros_ip`.
Can you please test this fix before I put it into the master branch?

NeelzDev, 2020-02-03:
Hey thanks for providing the update. 
Unfortunately, I still do not see any change.
I added the relevant output lines below.

As you mentioned before there should be some more output if a topic gets registered, e.g.: https://github.com/fkie/multimaster_fkie/blob/fix_sync_ros_ip/fkie_master_sync/src/fkie_master_sync/sync_thread.py#L324

I will try to add some more debugging output to understand better why it is not working. Maybe some check is failing due to a specific setup I have.

Thanks a lot for your support!

```
[INFO] [1580744045.079913]: ROS masters obtained from '/master_discovery/list_masters': ['10.42.0.133', '10.42.0.134']
[DEBUG] [1580744068.563584]: SyncThread[10.42.0.133]: update notify new timestamp(1580744068.482115984), old(1580744024.720185041)
[INFO] [1580744069.369647]: SyncThread[10.42.0.133] Requesting remote state from 'http://10.42.0.133:11611'
[INFO] [1580744069.379375]: SyncThread[10.42.0.133] Applying remote state...
[DEBUG] [1580744069.382525]: SyncThread[10.42.0.133]: remote state: ['1580744068.482115984', '1580744068.482115984', 'http://10.42.0.133:11311/', '10.42.0.133', [['/test', ['/rostopic_256_1580744068021']]], [], [], [['/test', 'std_msgs/String']], [['/rostopic_256_1580744068021', 'http://basic-pipeline-nils-ros-vnc:33963/', 'http://10.42.0.133:11311/', None, 'remote']], []]
[DEBUG] [1580744069.387647]: SyncThread[10.42.0.133]: current timestamp 1580744068.482115984, local 1580744068.482115984
[INFO] [1580744069.390684]: SyncThread[10.42.0.133] Requesting remote md5sums 'http://10.42.0.133:11611'
[INFO] [1580744069.396152]: SyncThread[10.42.0.133] remote state applied.
```

```
[INFO] [1580744069.055136]: SyncThread[10.42.0.134] Requesting remote state from 'http://10.42.0.134:11611'
[INFO] [1580744069.064085]: SyncThread[10.42.0.134] Applying remote state...
[DEBUG] [1580744069.067311]: SyncThread[10.42.0.134]: remote state: ['1580744015.036187887', '1580744015.036187887', 'http://10.42.0.134:11311/', '10.42.0.134', [], [], [], [], [], []]
[DEBUG] [1580744069.072681]: SyncThread[10.42.0.134]: current timestamp 1580744015.036187887, local 1580744015.036187887
[INFO] [1580744069.075639]: SyncThread[10.42.0.134] Requesting remote md5sums 'http://10.42.0.134:11611'
[INFO] [1580744069.081018]: SyncThread[10.42.0.134] remote state applied.
[DEBUG] [1580744084.199874]: connecting to 10.42.0.133 43161
```

atiderko, 2020-02-03:
thank you for testing!

The problem is in this entry:
`['/rostopic_256_1580744068021', 'http://basic-pipeline-nils-ros-vnc:33963/', 'http://10.42.0.133:11311/', None, 'remote']`

This is created by `master_discovery`. My fix [master_discovery/master_monitor.py#L817] (https://github.com/fkie/multimaster_fkie/blob/fix_sync_ros_ip/fkie_master_discovery/src/fkie_master_discovery/master_monitor.py#L817) and [master_discovery/common.py#L85](https://github.com/fkie/multimaster_fkie/blob/859138704885c1bd5e8e65a3251f82eee1b4608b/fkie_master_discovery/src/fkie_master_discovery/common.py#L85) should replace `basic-pipeline-nils-ros-vnc` by 10.42.0.133 if it is the *same host*! Remote nodes are not synchronized by master_sync. It could lead to problems...
If it is not the same host and you want synchronize remote nodes anyway, you have to set `sync_remote_nodes` of  `master_sync` to `True`.

NeelzDev, 2020-02-04:
I think the remote nodes is the issue in my case. Because both roscores are running on different pods (containers) with a separate network interface. 

Unfortunately `rosrun fkie_master_sync master_sync _log_level:=DEBUG sync_remote_nodes:=True`  or `rosrun fkie_master_sync master_sync _log_level:=DEBUG sync_remote_nodes:=true`  does not work. I added a debug output but the expression `self._filter.sync_remote_nodes()` always evaluates to false.
Unfortunately, I am far from being a ROS expert, I could not find the reason why the command option is not taken into account.

I disabled the check manually and now the topics get synched. The only thing I get is a warning in the discovery output "[WARN] [1580825992.122942]: 'NoneType' object has no attribute '_md5sum'".

You mentioned that there might be problems if remote hosts are synched:
What could be the problems if remote synch is used (what is unfortunately what I am looking for)?

Thanks a lot for your support!

atiderko, 2020-02-05:
1. `sync_remote_nodes` is local variable of `master_sync`, so you have prepend *_* and launch `master_sync` with:
`rosrun fkie_master_sync master_sync _log_level:=DEBUG _sync_remote_nodes:=True`
But you have changed the code and remote nodes are synchronized anyway ;-)

2. The warning is caused while `master_sync` checks the md5sum of remote topic with local one. I guess `master_discovery` can't resolve the topic class, since the node is not installed and not running in the same host/container as `master_dsicovery`.

3. (2.) is one of the problems with synchronization of remote nodes. Depending on ros-networks (multiple `master_sync` nodes) and timings the remote nodes could be stopped by roscore (multiple registration of nodes with same name.) It could also lead to visualization issues in `node_manager` do to dleays while tests for running state of the node.

You should start roscore, master_discovery and master_sync on each pod/container. This is the scenario what the multimaster was designed for.

I did not understand that the ROS nodes were started in another container than master_dsicovery. So the fix I did is for this problem not relevant.

Is there a reasen why you start the ROS-Nodes in different container as roscore and master_discovery?

NeelzDev, 2020-02-09:
Thanks. With setting it as local variable it works!

After all the setup is the following:
Inside the K8S cluster there is one pod for each ros program (node). Each of them is connecting to the ROS master(s). Because the ROS master is per default not backed up nor redundant, I was looking for running a redundant ROS master. Each instance of the ros master will run in their own pod. In that pod now the roscore, ros_master_sync and ros_master_discovery are running for each ros master. 
But all of the other nodes never know to which master they connect, as they are connecting via a service which randomly resolves to one of the ros masters. If old of them are in synch and one crashes for what reason so ever, it should be transparent to the nodes, as they will just register or request a topic from one of the other masters, which should by in synch.

It looks that it is running now, but apparently not 100% reliable. A colleague told me that he encountered some problems, apparently some topics have not been available. I am right now trying to reproduce the issue.

atiderko, 2020-06-18:
is the issue still present?

NeelzDev, 2020-07-05:
I will close it. 
There seem to be some reliability issues with the synch (sometimes topics don't show up or take a while to show up).

But overall it is working!

closed 2020-07-05 by NeelzDev