fkie/multimaster_fkie #120: multimaster_fkie on WSL

gstavrinos, 2020-02-25:
Hello,

I am trying to connect a WSL Ubuntu installation (https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux) on a multimaster system of 9 other rosmasters.

The problem I am facing is this:

All other (native linux) machines can discover the WSL machine, but the WSL machine cannot (it only detects itself) leading to a non-working system. No topics and services are synced.

All machines have the IPs and hostnames of the others in their hosts file, including the WSL one. The same goes for exporting the correct ROS_MASTER_URI.

Is there any suggestion or recommended way to run multimaster on a Windows machine?

Thanks in advance,

George

gstavrinos, 2020-02-25:
Update: `netstate -g` returns:
`no support for AF INET igmp on this system`

atiderko, 2020-02-26:
You can use `~robot_hosts` parameter to send unicast discovery messages to WSL machine.
e.g. on linux hosts:
```
rosrun fkie_master_discovery master_discovery _robot_hosts:=[wsl_machine_name]
```
or in launch file:
```
<rosparam param="robot_hosts">[wsl_machine_name]</rosparam>
```

gstavrinos, 2020-02-26:
Hey @atiderko, thanks for the quick reply. The problem with this is that I am afraid that it will flood the network and make it unusable. I will give it a shot though, just to check.

atiderko, 2020-02-26:
WSL does not see the linux machines its because WSL does not receive multicast messages from linux hosts. So `master_discovery` on linux should send unicast `heartbeat messages` to WSL. Therefore `robot_hosts` should be set on linux.

Because of flooding you can change the `heartbeat_hz` to reduce the send rate. You can set it also to zero. In this case heartbeats are send only on ROS-Master changes!

gstavrinos, 2020-02-26:
Woo! Didn't know that even with 0Hz heartbeat changes are caught! That was a great suggestion. I will get back when I try this and close the issue. Thanks, Alexander!

atiderko, 2020-02-26:
Good luck, George! I would be happy about a success message ;-)

gstavrinos, 2020-02-26:
Unicast did the trick! Off to the next steps! Thank you very much, Alexander!

closed 2020-02-26 by gstavrinos