fkie/multimaster_fkie #121: Writing a node that turns On/Off capabilities just like node_manager?

olamarre, 2020-02-26:
Hi! I've been using node_manager's Capabilities View to turn On/Off specific capability groups (it's great, thanks!).
Now, I would like to do the same but from a Python node (instead of using the node_manager GUI). Would you be able to advice which python package(s) & method(s) to use to accomplish this?
Thank you very much.

atiderko, 2020-02-27:
Hi!
you can call the ROS service `/node_manager_daemon/run` to run a node of a loaded launch file.
Load a launch file by itself is more complicated, see [launch_servicer.py](https://github.com/fkie/multimaster_fkie/blob/master/fkie_node_manager_daemon/src/fkie_node_manager_daemon/launch_servicer.py#L266) and [launcher.py](https://github.com/fkie/multimaster_fkie/blob/master/fkie_node_manager_daemon/src/fkie_node_manager_daemon/launcher.py#L62).

To stop a ROS node the [XMLRPC API](http://wiki.ros.org/ROS/Slave_API) of a ROS node is used, see [node manager code](https://github.com/fkie/multimaster_fkie/blob/master/fkie_node_manager/src/fkie_node_manager/master_view_proxy.py#L2215). For this call you need the XML-RPC URI of the node. You get this from ROS-Master XMLRPC by calling [lookupNode (Master API)](http://wiki.ros.org/action/fullsearch/ROS/Master_API#Name_service_and_system_state).

I hope this helps you achieve your goal.

olamarre, 2020-03-02:
Great, thank you for the detailed answer! I'll definitely refer to those to implement what I need.

closed 2020-06-18 by atiderko