fkie/multimaster_fkie #122: bash: rosrun: command not found

kledom, 2020-03-09:
Hello,

When running remote nodes I get the error 'bash: rosrun: command not found'. The troubleshooting section in the documentation suggest to remove the line `#[ -z "$PS1" ] && return `
from .bashrc. I think in my case (Ubuntu 18.04) I have to comment out the following line:
`case $- in
     *i*) ;;
       *) return;;
 esac`

However, I'm unsure on why this is necessary and what are the downsides of commenting it out?

atiderko, 2020-03-09:
Hi,

it is not necessary to comment it out. You can also put `source /opt/ros/melodic/setup.bash` before these lines.

kledom, 2020-03-09:
Thank you for your fast reply. Putting something before case $- in *i*) ;; *) return;; esac still has some major drawbacks.
To give an example:
I have a file ros_vars.bash that I source in .bashrc in order to setup ROS_IP and ROS_URI. I'm sure those variables are also required for node_manager. My script also outputs the current interface and ROS_IP on execution. 
If I put the line `source ~/ros_vars.bash ` before the non-interactive return statement, scp stops working because it can't handle the echo statements.

Could you change the rosrun command in node_manager so that bashrc is sourced beforehand? I think this would be a much cleaner solution.

atiderko, 2020-03-10:
if I source bashrc in a non interactive shell it exits at the same position `case $- in *i*) ;; *) return;; esac`.

Do you have a solution?

PS: remote nodes are started by [this code](https://github.com/fkie/multimaster_fkie/blob/master/fkie_node_manager/src/fkie_node_manager/start_handler.py#L150)

kledom, 2020-03-10:
I ended up adding the "case $- in *i*) ;; *) return;; esac" line to my ros_vars file and putting the ROS block at the top of .bashrc as you mentioned.

closed 2020-03-10 by kledom

atiderko, 2020-03-10:
:-/ I would have been happy to see a more elegant solution.
Thank you for posting your chosen solution!