fkie/multimaster_fkie #123: Python 3 compatibility

stertingen, 2020-04-12:
There are a few `except` usages, which are invalid for python 3:
https://github.com/fkie/multimaster_fkie/blob/master/fkie_master_sync/src/fkie_master_sync/master_sync.py#L137,
https://github.com/fkie/multimaster_fkie/blob/master/fkie_master_discovery/src/fkie_master_discovery/master_discovery.py#L937

Also,
will there be a python 3 compat update for the melodic branch (old master)? If yes, I might put some effort into a PR.

atiderko, 2020-04-21:
hi, sorry for delayed answer!

There are no plans to update the melodic branch to python 3. For Ubuntu 20.04 I plan to release the master branch with new node manager.

If needed I can add a new melodic python 3 branch and put in your PR.

stertingen, 2020-04-21:
I've started some work here: https://github.com/fkie/multimaster_fkie/pull/124

Any feedback is welcome.

atiderko, 2020-04-22:
Is there a reason why you did not use the new version in master branch?
The new version does not change the behavior of `master_discovery` and `master_sync`, apart from the renaming.
Furthermore a lot of code of the `node manager` is already python 3 compatible. And with your changes also `master_discovery` and `master_sync` would be python 3 compatible ;-)

stertingen, 2020-04-22:
> Is there a reason why you did not use the new version in master branch?
> The new version does not change the behavior of `master_discovery` and `master_sync`, apart from the renaming.

Well, there is no real reason, I thought the changes were bigger and incompatible to the old master, so I did not really look there. I will look into the new master branch now for inspiration.

> Furthermore a lot of code of the `node manager` is already python 3 compatible. And with your changes also `master_discovery` and `master_sync` would be python 3 compatible ;-)

Okay then, guess it won't be long until that PR is ready to merge.

atiderko, 2020-05-13:
Hi, I fixed (I hope all) python 3 compatibility problems for master branch.
Currently the package.xml file contains no python3 dependencies, because the pull requests for required dependencies in rosdep are currently pending.
But the code should be executable with python 3.

atiderko, 2020-09-18:
The current version is Python 3 compatible and runs with ROS noetic.
Please reopen on some problems!

closed 2020-09-18 by atiderko