fkie/multimaster_fkie #126: Ignore specific topics between two computers

Naraharirahul, 2020-05-13:
I'm trying to ignore a specific topic flowing from computer A to computer B. Is there any way of ignoring the topic apart from editing the computer B launch file?

atiderko, 2020-05-14:
Currently there is no way to configure master_discovery on computer A to not provide topics to master_sync on computer B. At this time you must configure master_sync.launch on computer B.

shervinkoushan, 2024-01-18:
@atiderko has anything been done in this area or is the status still the same?

atiderko, 2024-01-18:
yes, there are three parameters *~hide_nodes*, *~hide_topics* and *~hide_services* for master_discovery. This should prevent the *master_sync* on another host from seeing them. However, the Node Manager can still display them.

amilcarlucas, 2024-02-21:
Those parameters are not documented on http://wiki.ros.org/master_discovery_fkie?distro=melodic