fkie/multimaster_fkie #128: Is this hack still needed?

MJohnson459, 2020-05-29:
https://github.com/fkie/multimaster_fkie/blob/81a9d18c5100094b7f64abab249ed76b6383f41a/fkie_master_sync/src/fkie_master_sync/sync_thread.py#L435

Hi! I've been using FKIE with a large number of shared connections and I'm hitting an issue where if a new master connects, there is a huge flurry of publisherUpdates being performed. This in turn is causes issues by (what seems like) overloading some of the masters. This flurry is ultimately caused by the above code.

Looking at the ROS master API docs, it seems like this might not be required anymore. wiki.ros.org/ROS/Master_API

"Subscribe the caller to the specified topic. In addition to receiving a list of current publishers, the subscriber will also receive notifications of new publishers via the publisherUpdate API. "

I'm happy to supply a PR for the change but before I even tried removing it, I was hoping someone could help me understand why it exists in the first place.

atiderko, 2020-05-29:
Why the hack is there:
If we register a new publisher using 'registerPublisher' of wiki.ros.org/ROS/Master_API all subscriber are notified about [master_api.py#L758](https://github.com/ros/ros_comm/blob/9162b32a42b5569ae42a94aa6426aafcb63021ae/tools/rosmaster/src/rosmaster/master_api.py#L758)
On subscriber registration using 'registerSubscriber' we get only the list of subscriber  [master_api.py#L703](https://github.com/ros/ros_comm/blob/9162b32a42b5569ae42a94aa6426aafcb63021ae/tools/rosmaster/src/rosmaster/master_api.py#L703) and not publisher are notified. Therefore no connection between local publisher and remote subscriber will be established.

Perhaps it is possible to use undocumented [publisherUpdate](https://github.com/ros/ros_comm/blob/9162b32a42b5569ae42a94aa6426aafcb63021ae/tools/rosmaster/src/rosmaster/master_api.py#L195) to reduce the overloading of masters.

I'm happy about PR ;-)

atiderko, 2020-06-18:
replaced the hack by `publisherUpdate` call

closed 2020-06-18 by atiderko